workcraft/workcraft #113: Problem creating a new work while another work is maximised

workcraft-lp, 2016-02-07:
- Work with an existing document
- Maximize it's editor window
- Create a new document

Now it is not possible to exit the maximized editor mode properly....

The error message is:

Exception in thread "AWT-EventQueue-0" java.lang.IllegalStateException: Can't maximize while different dockable is maximized
	at org.flexdock.docking.DockingManager.toggleMaximized(DockingManager.java:2732)
	at org.workcraft.gui.MainWindow.toggleDockableWindowMaximized(MainWindow.java:448)
	at org.workcraft.gui.DockableWindowContentPanel$ViewAction.run(DockableWindowContentPanel.java:71)
	at org.workcraft.gui.MainWindow$1.actionPerformed(MainWindow.java:109)
	at org.workcraft.gui.tabs.TabButton.mouseClicked(TabButton.java:62)
	at java.awt.AWTEventMulticaster.mouseClicked(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)

workcraft-lp, 2016-02-07:
*danilovesky at 2014-09-04T15:11:23Z:*

After the editor panel is maximised and a new work is created I observe a different set of problems.
1) The two tabs have different styles - the initially maximised one has icons for exit the maximised mode while the second one has maximise button
2) If the maximised tab is closed, then the editor panel disappears completely (the second tab cannot be accessed any more)

danilovesky, 2016-07-27:
Another related problem reported:
Open an STG work and maximise its pane, then open another STG work and switch to the former STG.
Now both maximised and regular panes are visible.

danilovesky, 2017-04-26:
A solution is to un-maximise all panes on creation os a new work.

PR #676 (2017-04-26): Fixes for maximised tabs and editor focus
  * All tabs are un-maximised on creation of a new work.
  * Tabs are made non-focusable so they do not steal focus from
the ebmedded editor.
  * Focus is set to the editor on tab selection, header click,
maximisation and size restore.

Fixes #113
commit b3f511fe (2017-04-26): Fixes for maximised tabs and editor focus

  * All tabs are un-maximised on creation of a new work.
  * Tabs are made non-focusable so they do not steal focus from
the ebmedded editor.
  * Focus is set to the editor on tab selection, header click,
maximisation and size restore.

Fixes #113
closed 2017-04-26 by danilovesky