workcraft/workcraft #156: Platform shrinks to minimum size on MacOS

workcraft-lp, 2016-02-07:
Problem:
On MacOS, Workcraft is minimized and located on the bottom left of the screen after starting.
There is a warning message in the Terminal.
...
Switching to GUI mode...
WARNING: GL pipe is running in software mode (Renderer ID=0x1020400)
Now in GUI mode.

workcraft-lp, 2016-02-07:
*libowen96 at 2014-12-15T15:21:36Z:*

workcraft-lp, 2016-02-07:
*danilovesky at 2015-02-17T13:04:04Z:*

Jonny,if you can reproduce, could we debug this on your Macbook please?

workcraft-lp, 2016-02-07:
*jonathan-r-beaumont at 2015-02-18T12:49:46Z:*

I cannot reproduce this bug on my MacBook air.  
After starting, it produces the Workcraft window as expected in the top left corner, without any errors in the terminal.

danilovesky, 2016-02-29:
Could not be reproduced.

closed 2016-02-29 by danilovesky