workcraft/workcraft #198: Petri net conversion does not support pages in PolicyNet plugin

workcraft-lp, 2016-02-07:
To reproduce create a PolicyNet with two transitions t0 and t1. Put t0 into a page named node0. Select Tools->Conversion->Generate Petri net  to see the following exception:

Exception in thread "AWT-EventQueue-0" org.workcraft.exceptions.ArgumentException: "node0/t0" is not a valid C-style identifier.
The first character must be alphabetic or an underscore and the following characters must be alphanumeric or an underscore.
	at org.workcraft.dom.references.UniqueNameManager.setName(UniqueNameManager.java:81)
	at org.workcraft.dom.references.HierarchicalUniqueNameReferenceManager.setName(HierarchicalUniqueNameReferenceManager.java:204)
	at org.workcraft.dom.AbstractModel.setName(AbstractModel.java:195)
	at org.workcraft.plugins.petri.PetriNet.createTransition(PetriNet.java:104)
	at org.workcraft.plugins.petri.VisualPetriNet.createTransition(VisualPetriNet.java:79)
	at org.workcraft.plugins.policy.tools.PetriNetGenerator.convertTransitions(PetriNetGenerator.java:73)
	at org.workcraft.plugins.policy.tools.PetriNetGenerator.<init>(PetriNetGenerator.java:38)
	at org.workcraft.plugins.policy.tools.PetriNetGeneratorTool.run(PetriNetGeneratorTool.java:38)
	at org.workcraft.util.Tools.run(Tools.java:38)
	at org.workcraft.gui.MainWindow.runTool(MainWindow.java:1062)
	at org.workcraft.gui.MainMenu$ToolAction.run(MainMenu.java:68)
	at org.workcraft.gui.MainWindow$1.actionPerformed(MainWindow.java:108)
	at org.workcraft.gui.actions.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:58)
	at org.workcraft.gui.actions.ActionMenuItem.access$0(ActionMenuItem.java:55)
	at org.workcraft.gui.actions.ActionMenuItem$ActionForwarder.actionPerformed(ActionMenuItem.java:34)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2018)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2341)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6505)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3312)
	at java.awt.Component.processEvent(Component.java:6270)
	at java.awt.Container.processEvent(Container.java:2229)
	at java.awt.Component.dispatchEventImpl(Component.java:4861)
	at java.awt.Container.dispatchEventImpl(Container.java:2287)
	at java.awt.Component.dispatchEvent(Component.java:4687)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4832)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4492)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4422)
	at java.awt.Container.dispatchEventImpl(Container.java:2273)
	at java.awt.Window.dispatchEventImpl(Window.java:2719)
	at java.awt.Component.dispatchEvent(Component.java:4687)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:729)
	at java.awt.EventQueue.access$200(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:688)
	at java.awt.EventQueue$3.run(EventQueue.java:686)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.awt.EventQueue$4.run(EventQueue.java:702)
	at java.awt.EventQueue$4.run(EventQueue.java:700)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:699)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)

workcraft-lp, 2016-02-07:
*danilovesky at 2014-09-09T20:33:13Z:*

A temporary solution is to disable the Pages support in PolicyNet plugin.

danilovesky, 2020-10-11:
Policy net Localities conflict with Pages and therefore will not be implemented.

closed 2020-10-11 by danilovesky