workcraft/workcraft #84: Make the hazard reports more explicit

workcraft-lp, 2016-02-07:
The hazard reports only show the trace to the problematic state, but not the detail the signal that is actually hazardaous. This is confusing for non-experts.

The report should have the trace to the problematic state and somehow highlight the signal that is hazardous in that state, e.g. by drawing the bad pin or gate in red.

danilovesky, 2016-08-25:
The term "hazard" is too overloaded and our use of "hazard" does not match expectations of engineers. We need to replace it with "non-persistent" term, e.g. a message in the Verification result window should read:
```
Output persistency is violated

Non-persistency detected: INVdetouched+ is disabled by gp_ack+
<TRACE> [PLAY BUTTON]

Non-persistency detected: aaa+/3 is disabled by bbb-/2, bbb-/2 is disabled by aaa+/3
<TRACE> [PLAY BUTTON]

...
```

danilovesky, 2016-08-25:
If several non-persistent signals are detected at the reached state, then replicate the trace and append disabling transitions to show all possibilities. E.g. the following message:
```
Output persistency is violated

Non-persistency detected: aaa+/3 is disabled by bbb-/2, bbb-/2 is disabled by aaa+/3
a1+, a2-, a3+
```
would translate into the following two traces:
```
Output persistency is violated

Non-persistent signal aaa (aaa+/3 is disabled by bbb-/2)
a1+, a2-, a3+, bbb-/2

Non-persistent signal bbb (bbb-/2 is disabled by aaa+/3)
a1+, a2-, a3+, aaa+/3
```

closed 2016-09-19 by danilovesky