workcraft/workcraft #328: Replace jars with imports from jcenter/mavenCentral where possible

mvdan, 2016-02-08:
To reduce the repo HEAD size and also simplify the maintenance of the libraries (keeping them up to date, etc).

mechkg, 2016-02-08:
Below is what I remember from 4 years ago, might be obsolete.

Substance UI library has been discontinued. https://github.com/Insubstantial/insubstantial is a maintained fork. 

Flexdock is not in any centralised Maven repo. We used to have a local repo server where we installed it manually. It is released under a permissive license though, so might make sense to include it in the Workcraft's source tree instead.

Everything else is fairly popular and should be available from public Maven repos.

mvdan, 2016-02-10:
Got rid of 5 jars so far. Looks like quite a lot of the libs are in neither of jcenter/mavenCentral. I'd rather keep them in the repo as jars than set up our own maven repository.

mvdan, 2016-02-10:
Only three are left:

* jbpt - can't find it anywhere
* flexdock - can't find it anywhere
* batik - we use 1.8, mavenCentral/jcenter only have 1.6 for some reason

mvdan, 2016-02-11:
Replaced everything that I could, closing. The remaining can be gradually removed.

closed 2016-02-11 by mvdan