workcraft/workcraft #352: Create a git repo holding the tool binaries

mvdan, 2016-02-15:
To make it possible for everyone to create release tarballs, we should publish the stuff missing from this repo (`distr-template-linux` et al).

Adding it to this repo would make it unnecessarily larger, so let's use a separate repo like `tuura/workcraft-backend-bins` and link to it via git submodules.

danilovesky, 2016-02-15:
Wouldn't it be better to have two repos, say `tuura/workcraft-backend-windows` and `tuura/workcraft-backend-linux` ?

mvdan, 2016-02-15:
Either one repo with two folders or two repos, it really doesn't make any difference. A single repo would make more sense in the way that whenever a tool is updated, only one repo is updated and only one submodule needs to be updated.

danilovesky, 2016-02-15:
A repository is created with `windows` and `linux` directories: https://github.com/tuura/workcraft-distr-template

closed 2016-02-15 by danilovesky