workcraft/workcraft #378: Improve dist.sh script

danilovesky, 2016-02-25:
Requirements:
  * Both Windows and Linux distros should be created under dist/windows and diest/linux respectively. No PLATFORM parameter is required.
  * An optional TAG parameter should be possible. If it is omitted than `git describe --tags` should be used as the TAG.
  * The compressed files should be `workcraft-TAG.zip` and `workcraft-TAG.tar.gz` for Windows and Linux respectively. 
  * The top directory inside the `zip` and `tar.gz` should we `workcraft-TAG`.

mvdan, 2016-02-25:
> Both Windows and Linux distros should be created under dist/windows and diest/linux respectively. No PLATFORM parameter is required.

If no platform parameter is required, how do you know which one you're building?

> An optional TAG parameter should be possible. If it is omitted than git describe --tags should be used as the TAG.

Why? For what purpose?

> The compressed files should be workcraft-TAG.zip and workcraft-TAG.tar.gz for Windows and Linux respectively.

Having the platform as part of the filename is better. zip is not limited to Windows and tar.gz is not limited to Linux. But their contents are.

> The top directory inside the zip and tar.gz should we workcraft-TAG

They should have the same name as the archive, so whatever we do in the point above is what we'll do here.

danilovesky, 2016-02-25:
Which to build? - Always both.
Why? - For (my) convenience.
windows/linux suffix in the zip/tar.gz - Agree.
They should have the same name - Disagree. The end user knows what OS he is running, no need to remind this in the directory name.

mvdan, 2016-02-25:
> Always both.

What if you just want to build one?

> For (my) convenience.

Sure, feel free to add an option.

> Disagree. The end user knows what OS he is running, no need to remind this in the directory name.

It is common and encouraged to use the same archive name as the directory name. `foo.tar.gz` should decompress into `foo/`.

mvdan, 2016-02-25:
As for the platform suffix, it's already there.

danilovesky, 2016-02-25:
For distribution I always need two. One way to get one - build two and remove one of them.

I do not want the platform suffix in the directory name. After unpacking it should be just `workcraft-TAG`. If it results in removing the suffix from the archived file - so be it.

mvdan, 2016-02-25:
So if I just want to build one, I have to waste resources and time building all of them and then ignore the ones I don't need?

Why don't you want the platform suffix in the directory name?

danilovesky, 2016-02-25:
That is one of the possibilities.
As above: The end user knows what OS he is running, no need to remind this in the directory name.

mvdan, 2016-02-25:
> That is one of the possibilities.

My point is that it should be possible to just build one. If you want an easier way to build all of them, then let's add an `all` target.

> As above: The end user knows what OS he is running, no need to remind this in the directory name.

So having it in the archive name is fine, but in the folder name is unacceptable? I don't follow your logic.

danilovesky, 2016-02-25:
I want both platforms to be built as the default behaviour. And the user-defined tag should have priority over platform in the command line. That is the most often use case for me. If there is a way to add an option to build a single platform, e.g. `--platform PLATFORM` parameter I am ok with that.

I would prefer not to have platform suffix in the file name. But can live with that.

mvdan, 2016-02-25:
> I want both platforms to be built as the default behaviour. And the user-defined tag should have priority over platform in the command line. That is the most often use case for me. If there is a way to add an option to build a single platform, e.g. --platform PLATFORM parameter I am ok with that.

I can work with that.

I'll post a PR soon.

PR #392 (2016-03-02): Dist script improvements
Fixes #378.
closed 2016-03-02 by danilovesky