workcraft/workcraft #379: Pass plugins directory as a command line parameter

danilovesky, 2016-02-25:
Currently CLASSPATH environment variable is used to scan for plugins.
This does not work well with CLASSPATH set in JAR files.
The current workaround is to add all JARS in the CLASSPATH and explicitly point to `org.workcraft.Console`:
```
export CLASSPATH="$WORKCRAFT_HOME/workcraft.jar:$WORKCRAFT_HOME/plugins/*"
"$JAVA_BIN" org.workcraft.Console -dir:"$CURRENT_DIR" "$@"
```
A better solution would be to pass the plugins directory as a`-plugins:` parameter, something like this:
```
"$JAVA_BIN" -jar workcraft.jar -plugins:"$PLUGINS_DIR" -dir:"$CURRENT_DIR" "$@"
```

mvdan, 2016-02-25:
Why are you using this `-option:value` format? `-option=value` is a lot more common.

danilovesky, 2016-03-05:
Historically the command line options are passed like this. I do not have strong preferences on the format and follow this one for consistency.

danilovesky, 2020-03-26:
All JAR files in ``bin`` directory of the distribution are added to the CLASSPTAS:

    export CLASSPATH="$WORKCRAFT_HOME/bin/*"

And all the necessary third-party libraries in the ``lib`` directory of the distribution are added as relative paths in ``MANIFEST.MF`` file of the corresponding JAR file, e.g:

    Class-Path: ../lib/jdd-111.jar ../lib/flexdock-1.2.4.jar

This is closed by PR #1084

closed 2020-03-26 by danilovesky