workcraft/workcraft #413: 3.0.9 Error: draw State graph

ehnatvisiontracedotdk, 2016-03-07:
Petrify tool chain execution failed :-(

Failure caused by:
Errors running draw_astg:
sh: 1: dot: not found
*** FATAL ERROR: dot could not be executed successfully

mvdan, 2016-03-07:
Did you try installing `dot`?

Perhaps the error on our end should be more descriptive, telling users to go install it.

ehnatvisiontracedotdk, 2016-03-07:
Resolved.

danilovesky, 2016-03-07:
For Windows version we include a cut-down version of Graphviz. The problem is that one of Petrify tools (draw_astg) calls `dot` too.. .
Could you try this internal release please (without installing dot) - we tried to fix it there:
https://drive.google.com/file/d/0B2s4C-vvCyaidzV4ZUJCZWpnZG8/view?usp=sharing

closed 2016-03-07 by danilovesky