workcraft/workcraft #416: Verification and synthesis problems with Ubuntu on VirtualBox

dsanz006, 2016-03-08:
I am using Workcraft 3.0.9 in a Mac computer, so I use VirtualBox with an Ubuntu virtual machine. 
Workcraft seems to open correctly but the verification and synthesis tools seem not to work.
On Ubuntu 15.10 the program halts when doing any verification. 
I have also tried previous versions of Ubuntu, and verification works correctly. But when trying to synthesise the program does not do anything. In the problems tab, I can see the following: 
[image]

danilovesky, 2016-03-08:
It looks like `petrify` tool could not even start. Could you report the output of `ls -l tools/PetrifyTools/` command in your Workcraft directory please?

dsanz006, 2016-03-08:
Thanks for your answer. The output of that command is the following: 
[image]

danilovesky, 2016-03-08:
Thanks for the report! All looks good in `tools/PetrifyTools/`...

Just to narrow things down, is it Ubuntu 15.10 or 15.04 you are currently experimenting with? I usually stick to LTS versions, so still on Ubuntu 14.04, but I will install the same version as you to catch the bug. 

Could you install v3.0.9-rc2 please from here:
https://drive.google.com/file/d/0B2s4C-vvCyaiUjFGZUxONDFmOTg/view?usp=sharing

and  copy-paste the log from `Output` tab of Workcraft just after an unsuccessful `petrify` call (e.g. via `Synthesis->Complex gate [Petrify]`. There should be a line starting with `[INFO] Running external command:` -- that message is of particular interest.

dsanz006, 2016-03-08:
My Ubuntu version is 15.10. I had exactly the same problem with Ubuntu 14.04, and I upgraded thinking that it would be solved, but it did not. I think it must be related to using VirtualBox rather than to the Ubuntu version. 
I have downloaded the version you gave me and the problem remains. The log from the `Output` tab of Workcraft, just after `Synthesis -> Complex gate [Petrify]` is the following: 

`Exporting model 'problem_13_c' to file '/tmp/workcraft-1721399522942140617/petrify.g'.
[INFO] Running external command: /home/dsanz006/Downloads/workcraft/tools/PetrifyTools/petrify -cg -o /tmp/workcraft-1721399522942140617/result.g -eqn /tmp/workcraft-1721399522942140617/petrify.eqn -vl /tmp/workcraft-1721399522942140617/petrify.v -blif /tmp/workcraft-1721399522942140617/petrify.blif -log /tmp/workcraft-1721399522942140617/petrify.log /tmp/workcraft-1721399522942140617/petrify.g`

The log on the `Problems` tab seems to be the same as before.

danilovesky, 2016-03-08:
Most likely you are right and it is something related to VirtualBox... Do you run 64-bit or 32-bit distribution of Ubuntu?

Could you do one more thing please? Do `Synthesis->Complex gate [Petrify]` again and while Workcraft is still running (so all the temporary files are still there) execute the same command (the long line after `[INFO] Running external command:`) from a terminal? Then copy-paste the whole output you get in the terminal window.

dsanz006, 2016-03-08:
It is the 64-bit version.
What I get after executing the command from a terminal is this:

`bash: /home/dsanz006/Downloads/workcraft/tools/PetrifyTools/petrify: No such file or directory
`
And that is weird because I can see the `petrify` object if I change directory to ..../PetrifyTools/ on a terminal window. How is it possible that I can see it but bash reports that there is no such file or directory? Perhaps this is the reason for the problem?

danilovesky, 2016-03-08:
Yes, this is exactly the reason...

So, if you do `ls -l /home/dsanz006/Downloads/workcraft/tools/PetrifyTools/petrify` (from anywhere) do you see a binary with executable attributes for everybody? Something like this:
```
-rwxrwxr-x  dsanz006  dsanz006  1705204  Mar 8  21:55  petrify
```

dsanz006, 2016-03-08:
The output of the command looks exactly as you wrote: 

`-rwxrwxr-x 1 dsanz006 dsanz006 1705204 Mar  6 21:05 /home/dsanz006/Downloads/workcraft/tools/PetrifyTools/petrify`

When I do `pwd`, this is the path that I get:

`dsanz006@dsanz006-VirtualBox:~/Downloads/workcraft/tools/PetrifyTools$ pwd
/home/dsanz006/Downloads/workcraft/tools/PetrifyTools`

danilovesky, 2016-03-09:
@dsanz006: Looks like the problem is that Petrify is a dynamically linked 32-bit library and does not work in a vanilla 64-bit Ubuntu. Some 32-bit compatibility libraries need to be installed, see this discussion for details:

http://askubuntu.com/questions/454253/how-to-run-32-bit-app-in-ubuntu-64-bit

The error message is very confusing...

A proper solution would be to statically link Petrify so it does not depend on (not) installed libraries.

dsanz006, 2016-03-09:
I have done what the link you send me suggests, and now the program halts completely when I try to synthesize (before it didn't synthesize but at least I could continue running it). 
Would it be a solution to create a new virtual machine using 32-bit Ubuntu? can I even do that on my 64-bit mac machine?

danilovesky, 2016-03-09:
Strange.. Before trying 32-bit Ubuntu or going for statically linked Petrify could you run the whole petrify command in terminal please? Copy-paste it into terminal from the Output tab while Workcraft is still running, as you did yesterday

dsanz006, 2016-03-10:
I think we are not in the same uni, I am in the Technical University of Denmark...

danilovesky, 2016-03-10:
I see...

A proper solution has two non-trivial steps (will take time):

1) Include statically linked binaries for backend tools in Linux distribution of Workcraft. For this we need to contact the authors of the tools (as not all the sources are easily available). 

2) Compile all the backend tools for OSX and release a Mac distribution of Workcraft. For this we need to source a macbook and find some free time.

As a temporary solution I will try to borrow a macbook tomorrow and try to reproduce your problem locally. Then I will be in a better position to estimate if a quick fix can be made.

dsanz006, 2016-03-10:
Thanks! it would be great if you could reproduce my problem, because the 2 steps you mention seem like they will take a long time.

danilovesky, 2016-03-11:
I managed to reproduce your problem in the VirtualBox and Ubuntu 15.10. The last problem you face (freezing of th GUI) is most likely due to OpenJDK incorrectly handling external processes. A quick way to check it is to install Oracle JVM.  

In general there are 3 things one needs to be aware of while running Workcraft:
  * Disk size of at least 10GB (better make it 12GB to be safe) - this modern Ubuntus are very hungry for space. Alternatively you can clean up the unnecessary packages after installation. 
  * 32-bit compatibility libraries for Petrify.
  * Version of JVM - I had problems with OpenJDK, but not with Oracle.

Here is step-by-step instructions of a successful installation:

1. Install VirtualBox (I used v4.3.36) and Guest Additions (optional).

2. Create virtual machine:
  * HDD: 12GB (at least 10GB is required)
  * RAM: 4GB (may be less, e.g. 1GB)
  * Video: 32MB (can be less, e.g. 12MB)

3. Install Ubuntu 15.10 64-bit:
  * Download updates while installing  - Yes
  * Install this third-party software  - Yes

4. (Optional) After the reboot update via *Software Upader*.

5. (Optional) After  reboot install guest OS additions
  (*Devices->Insert Guest Additions CD image...*). This is
  necessary for better display drivers and shared folders.

6. Install 32-bit compatibility libraries:
   `$ sudo apt-get install libc6:i386`

7. Install Oracle JVM (note that OpenJDK gaves problems on execution of backend tools):
```
  $ sudo add-apt-repository ppa:webupd8team/java
  $ sudo apt-get update
  $ sudo apt-get install oracle-java8-installer
```

Finally download and install Workcraft v3.0.9-rc3 (previous versions should also work, but this one is the latest internal release): https://drive.google.com/file/d/0B2s4C-vvCyaiOURjWjRVM3dBejA/view?usp=sharing

dsanz006, 2016-03-13:
SOLVED! I did not have to create a new virtual machine because I had the space requirements in the one I already had. What I did was to re-install the 32-bit compatibility libraries and to install Oracle JVM (steps 6 and 7 in your instructions). Then I installed workcraft v3.0.9-rc3 and both verification and synthesis work.
Thank you very much! How can I mark this thread as solved?

danilovesky, 2016-03-14:
I think we better keep this issue open until a statically linked Petrify is bundled with Workcraft. The above solution is just a quick a workaround. Also we need to study why OpenJDK does not work well with external processes under VirtualBox (but works ok in a couple other Linux machines that we tried).

commit 9111424f (2016-03-22): Update dist-template submodule

It contains a statically linked Linux binary of Petrify tool.

Closes #416.
closed 2016-03-22 by danilovesky