workcraft/workcraft #417: workcraft-v3.0.9-beta Drawing Error - Synthesis: generalised C-element[MPSat]

fabigB, 2016-03-08:
Hello, I synthesised the attached file using the generalised C-element[MPSat] option and I believe that there should be an OR in the drawing instead of a c-element.
Otherwise create tool. Thanks!
[DrawError.zip](https://github.com/tuura/workcraft/files/163924/DrawError.zip)

Regards
Fabian

danilovesky, 2016-03-08:
Thanks for the report Fabian!

I think you confused the log from Petrify (equations in the screenshot) and the circuit from MPSat (schematic in the screenshot). Most likely you did generalised C-element synthesis with Petrify and then with MPSat and scrolled the Output log up too much...  

MPSat generates the following generalised C-element implementation, which corresponds to the cisrcuit in your screenshot:

```verilog
module MPSAT_unmapped_implementation(xack, yack, zreq, xreq, yreq, zack);
    output xack, yack, zreq;
    input xreq, yreq, zack;

    // gC implementation, each 'assign' statement is assumed to be ATOMIC;
    // total cost=10

    // SET=xreq & zack, RESET=~zack, cost=3
    assign xack = xreq & zack | xack & zack;
    // SET=~yreq & zack & ~yack, RESET=~zack, cost=4
    // assign xack = ~yreq & zack & ~yack | xack & zack;

    // SET=yreq & zack, RESET=~zack, cost=3
    assign yack = yreq & zack | yack & zack;
    // SET=~xreq & zack & ~xack, RESET=~zack, cost=4
    // assign yack = ~xreq & zack & ~xack | yack & zack;

    // SET=yreq | xreq, RESET=~xreq & ~yreq, cost=4
    assign zreq = yreq | xreq | zreq & (xreq | yreq);

    // initial values of the signals
    // !xreq !yreq !zack !xack !yack !zreq
endmodule
```

Please let me know if I am missing something...

danilovesky, 2016-03-08:
I agree that an OR-gate would be a better choice for `zreq` signal though... It may be the case that in this synthesis mode MPSat intentionally chooses a larger implementation to make sure all the signals are implemented as C-elements. 

I will pass this benchmark to the author of MPSat.

vkhomenko, 2016-03-09:
I believe MPSat does the right thing:

- gC synthesis simply derives the SET and RESET covers - these are shown in the comments in the Verilog file and my understanding is that nobody question their correctness.
- **These covers are meant to be implemented by pull-up and pull-down networks of transistors, not by logic gates.**
- These covers have then to be passed to the front-end. For some strange reasons Verilog is used for this, and its `assign` statements are abused for this purpose. Since both SET and RESET covers have to be passed for the same signal, they are combined into `assign z = SET | (z & RESET)` for this.
- The generated Verilog thus is not to be taken as true Verilog, but rather as some intermediate representation of the covers which must be in Verilog format but has nothing to do with actual Verilog. This Verilog file is only meant for the developers, not for people.
- The front-end reads this file and recovers the SET and RESET functions for each signal from each assign statement: `SET=assign_body[z=0]` and `RESET=~assign_body[z=1]`.
- Then front-end creates a gC element with these covers. It is not clear that it must try to match it to any logic gate - it's not a trivial problem, and the user might want to see these covers rather than logic gates. In any case, this has nothing to do with MPSat at this stage.

danilovesky, 2016-03-09:
@vkhomenko, do you think it worth spending time on improving the rendering function in Workcraft so it recognise complimentary set/reset functions of a component and visualise it as a combinational gate?

vkhomenko, 2016-03-09:
Actually I have now had a look at the screenshot - it looks like this is a front-end bug after all - it would be ok to draw a gC element with the correct covers for zreq, but WC draws a C-element instead! So it does try to recognise the gate, but goes wrong in this case for some reason.

Generally, for gC synthesis I don't see any need to recognise gates, as the signals are meant to be implemented by transistor networks rather than logic gates. The other synthesis modes (CG, stdC and tech mapping) cater for gates.

danilovesky, 2016-03-09:
Ok, will force BOX rendering for this synthesis mode then.

vkhomenko, 2016-03-09:
Well, the recognition bug still needs fixing - it may pop up somewhere else (e.g. when the user creates the circuit by hand and inputs these SET and RESET functions). BTW, the reason could be that the front-end does not allow negations anywhere except at the level of literals, but the reset function is restored by it as ~(xreq | yreq); entering it as ~xreq & ~yreq instead might fix the issue. If so, you will need to push the negations to literals when restoring SET / RESET functions or when the user enters them.

fabigB, 2016-03-09:
Hello,
I did the generalised C-element synthesis with Petrify first. I did not notice that I scrolled up to much. Thanks for your response!

danilovesky, 2016-03-09:
I rely on @snowleopard help to improve the gate rendering code.

snowleopard, 2016-03-09:
@danilovesky Let's formulate the precise conditions for when a gate should be rendered as a C-element and how to break ties (e.g. prefer less inversions in pins).

Fixing the code should be relatively straightforward after that.

snowleopard, 2016-03-09:
The case with the usual symmetric C-element seems easy: the `set` function should be equivalent to a conjunction of literals, and the `reset` function must have the same set of literals but with inverted polarity.

vkhomenko, 2016-03-09:
yes, and to check that you need to check that one of the co-factors for each variable is equivalent to 0, i.e. the problem is equivalent to UNSAT.

To check if SET/RESET result in a combinational gate, one has to check SET=~RESET, i.e. SET <=> RESET is UNSAT.

Has the SAT / UNSAT check been implemented in the front-end? 

Are any other gates besides C and combinational ever rendered?

snowleopard, 2016-03-09:
@vkhomenko I guess we can permit clever rendering only for gates with, say, up to 16 inputs, for which we can implement the check naively, without relying on proper SAT/UNSAT solvers.

> Are any other gates besides C and combinational ever rendered?

We do attempt to correctly draw asymmetric C-elements. The condition here seems to be more involved (instead of equality of literal-sets, we need a subset relation?).

vkhomenko, 2016-03-09:
Everything besides combinational gates, C elements, and boxes can be rendered as asym-C, can't it?

snowleopard, 2016-03-09:
@vkhomenko Sounds like an interesting conjecture, not sure if it's true :)

vkhomenko, 2016-03-10:
This is rather enigmatic - can you explain? There are no other rendering types in WC yet, right?

snowleopard, 2016-03-10:
@vkhomenko Well, there are plain boxes. If we don't know how to render a particular gate because it has strange set/reset functions, we can always fall back to using a box instead.

You said that everything is either combinational, C element, or an asymmetric C-element. That doesn't sound like a trivial statement to me.

vkhomenko, 2016-03-10:
I also said "and boxes" ;-)
How do you recognise whether given SET / REST functions form asym-C? What would be a general definition of an asym-C?

snowleopard, 2016-03-10:
@vkhomenko I think asym-C is a C-element, which also has asymmetric sets `Plus` and `Minus` that are only active in the respective phase. That is `set = f(S) /\ g(Plus)`, `reset = ~f(S) /\ h(Minus)`, where `S` is the common set of literals as in C-element, and `Plus U Minus` is not empty. Does this sound right?

vkhomenko, 2016-03-10:
Can Plus and Minus overlap? Can Plus and/or Minus overlap with S?

snowleopard, 2016-03-10:
@vkhomenko I think no to both questions: `S`, `Plus` and `Minus` do not overlap.

vkhomenko, 2016-03-10:
Ok, my previous question is irrelevant - I think I can prove that any SET / RESET can be implemented by asymC. Suppose f(S)=1 and we create new fresh disjoint sets of variables Plus and Minus corresponding to the supports of SET / RESET respectively. Then we just connect the same wire to one or more pins of the resulting asymC. Makes sense?

snowleopard, 2016-03-10:
Looks reasonable. Of course we would want to have as large `S` as possible for clearer and canonical representation.

danilovesky, 2021-09-28:
Let us stick with this definition of asymmetric C-element: https://en.wikipedia.org/wiki/Asymmetric_C-element

Here the interpretation of _Common/Plus/Minus sets_ is as follows:
* C-element goes high when all its inputs in _Common set_ and _Plus set_ are high
* C-element goes low when all its inputs in _Common set_ and _Minus set_ are low

With this definition in mind, it would not be possible to represent an arbitrary sequential gate as an asymmetric C-element. Let us consider a set-dominant C-element with a S that forces C-element high when S=1, does not matter what other inputs are. Its set/rest functions are as follows:
`set = S + A * B`
`reset=!S * !A * !B`

I cannot figure out _Common / Plus / Minus sets_ for these equations that would satisfy the above definition of asymmetric C-element.

snowleopard, 2021-09-28:
@danilovesky If I understand Victor's idea correctly, you can rewrite `set` as `(S + A)(S + B)` and then implement this gate as follows: 

* The Common set is empty.
* The Minus set is `{S, A, B}`.
* The Plus set is `{X, Y}` where the two fresh variables are implemented as `X = S + A` and `Y = S + B`.

Not sure if we'd like to support such "complex" asymmetric C-elements though. I agree that this gate seems to be unimplementable without introducing fresh variables.

vkhomenko, 2021-09-29:
This definition essentially means that for asymC elements without input or output bubbles, the set function is a product of positive literals, and the reset function is a product of negative literals (with the variables in common forming the common set). This is more restrictive than Andrey's definition above and so my idea won't work - it assumes arbitrary pull-up/pull-down networks are allowed, not just vertical stacks.

If we take Danil's definition of asymC, should we have a separate definition for gC? Or just represent the latter as a box with set/reset functions?

danilovesky, 2021-09-29:
I am not sure if we should stick to the "classical" definition of asymmetric C-element that only allow PMOS and NMOS transistors connected in series. I definitely saw an "extension" of this with input bubbles in the Plus/Common/Minus sets, so I would probably go for rendering those too.

Should we allow arbitrary transistor stacks? I am not sure either. I guess it may be possible by "attaching" OR/AND/XOR symbols in the Plus/Common/Minus sections of the C-element symbol/ This may look unconventional though.

@vkhomenko We do already render gC gates as Boxes, at least when they come from MPSat/Petrify.

vkhomenko, 2021-09-29:
I'd allow input/output bubbles to correct the phases of literals. If we go for arbitrary gC elements, we could draw that much like multi-level complex-gates, just with an asymC gate at the top.

vkhomenko, 2021-09-29:
Given the availability of BDDs at the front-end, we could do better recognition of gates, i.e. semantic rather than syntactic. E.g.:
 
-  If SET <-> ~RESET then the gate is combinational even though RESET is present. 
-  If both SET and RESET are cubes (which can be checked using BDDs - e.g. for each variable one of the two co-factors is 0, but this probably can be improved using low-level BDD operations - essentially BDD is a list rather than tree, so one of the pointers at each node must point to 0 and the other pointer is the tail of the list) and RESET cube can be obtained from the SET cube by flipping all inversions then it's a C-element (perhaps with bubbles).
- If both SET and RESET are cubes and each common variable x occurs in these two cubes in opposite phases than it's asymC.

closed 2021-10-01 by danilovesky

snowleopard, 2021-10-02:
@danilovesky Great to see this issue fixed :) What version did you end up implementing?

danilovesky, 2021-10-02:
@snowleopard I leaned towards my variant (https://github.com/workcraft/workcraft/issues/417#issuecomment-930100333). It fits the existing "codestyle" of using Set/Reset functions in few other places (I may revisit it later). Anyway, it is definitely an good step forward.

snowleopard, 2021-10-02:
@danilovesky Thanks, sounds good!

commit faf40999 (2021-10-01): Improve rendering of sequential gates

Closes #417