workcraft/workcraft #418: No way to fetch running Workcraft version

mvdan, 2016-03-08:
Say you're a user on a machine that isn't yours that has Workcraft. How do you tell which version you're running?

As far as I know, there currently isn't a way. The string `3.0.` isn't anywhere in the repo. We should have a `-version` for the cli interface, and if possible have the version included in an "about" dialog or window.

mvdan, 2016-03-08:
FWIW, this came to mind while writing the issue reporting template.

danilovesky, 2016-03-08:
Actually there is a version number in the About dialog. It is also reported to `stdout` as the first line after Workcraft start.

Note that the version is converted into a string in `Info.java` class -- that is why the string `3.0.` cannot be found.

I agree that the `-version` parameter would be useful.

mvdan, 2016-03-08:
Whoops, my bad then. Let's leave this open for the flag then.

danilovesky, 2016-03-09:
`-help` would be also useful.

commit 1e0e7dd7 (2016-03-10): Added -version and -help command line parameters

Closes #418.
closed 2016-03-10 by danilovesky