workcraft/workcraft #421: Workcraft distribution for OS X

danilovesky, 2016-03-12:
Java frontend runs fine on OS X, provided Oracle JVM 7 (or newer) is installed.

OS X binaries are needed for all the backend tools:
* PetrifyTools - binaries exist https://www.cs.upc.edu/~jordicf/petrify/distrib/petrify-4.2-mac.tgz
* Graphviz - binaries exist http://www.graphviz.org/Download_macos.php
* Espresso - needs building https://embedded.eecs.berkeley.edu/pubs/downloads/espresso/index.htm
* ScEnco - needs building https://github.com/tuura/scenco
* PGMiner - needs building https://github.com/tuura/process-mining
* UnfoldingTools - sources can be obtained, needs building
* vxm - sources can be obtained, needs building

martinjlowm, 2016-05-19:
I would like to contribute a Homebrew formula to install Workcraft on OS X, but I am missing sources for UnfoldingTools and vxm.

What is the arrangement with Victor Khomenko and his UnfoldingTools? Does one have to request the sources from him?

What about vxm? @danilovesky, I see you are co-author of the "XMASCRAFT: a tool for GALS Modelling and Verification of Communication Circuits" paper so I presume you can answer this.

I apologise if this isn't the right place to ask questions. I couldn't find any information regarding a mailing list or IRC channel.

danilovesky, 2016-05-20:
@martinjlowm: Thanks for offering help!

Is it ok for Homebrew to operate on binaries or one needs source code too? The problem is that for various reasons some of the tools are not really open-source (e.g. UnfoldingTools and PetrifyTools). Well, it is the choice the authors made...

We are currently building all the backend tools for mac and hope to have binaries in the next few days. Petrify (v5.0) and UnfoldingTool are the most difficult ones, but we are almost there. All the backend binaries will be in [workcraft-dist-template](https://github.com/tuura/workcraft-dist-template) under `mac` or `osx` directory (please advise what is a better name as here).

About VXM - the author is Frank Burns (I did the Workcraft side of xMAS mostly). I am currently chasing him to get the latest version of the sources :)

danilovesky, 2016-05-20:
@martinjlowm If you are experienced in building legacy code for Mac, could you give Espresso a try please? It did not work out smoothly, so the code will need some massaging. The source code is here: https://embedded.eecs.berkeley.edu/pubs/downloads/espresso/espresso.tar.gz

32-bit build would be preferable as there may be some pointer arithmetic that relies on this.

danilovesky, 2016-05-20:
@martinjlowm We just managed to compile `espresso`, please do not waist your time on this :)

martinjlowm, 2016-05-20:
> @martinjlowm: Thanks for offering help!

No problem! I've been using Workcraft lately as part of an asynchronous circuits course and so far I have X forwarded the application from a virtual machine. It works but is not the most optimal experience. I thought I might as well make it easy for newcomers that use a Mac to install and use the program.

> Is it ok for Homebrew to operate on binaries or one needs source code too? The problem is that for various reasons some of the tools are not really open-source (e.g. UnfoldingTools and PetrifyTools). Well, it is the choice the authors made...

I think you are right about this. However, Homebrew supports self-hosted taps that a user can subscribe to and these formulae do not have to be merged into the official Homebrew formula repository. A simple use case is the following assuming I am hosting the formulae here on Github:

`brew tap martinjlowm/workcraft-brew`
`brew update`
`brew install xxx`

As for binaries, there is the Cask supplement to Homebrew (https://github.com/caskroom/homebrew-cask).

> We are currently building all the backend tools for mac and hope to have binaries in the next few days. Petrify (v5.0) and UnfoldingTool are the most difficult ones, but we are almost there. All the [...]

That is great! I will wait a bit longer and watch out for activity here. I would advise you to name the directory osx. A Hackintosh should (of course) be able to run the software as well ;)

danilovesky, 2016-05-24:
@jrbeaumont: I will do the `createDialogButton` thing. Please have look at the mac-specific shortcuts.

jrbeaumont, 2016-05-24:
I'll test meta on Windows, Linux and OS X and see if there's any issues. If not, Toolkit works.  

I noticed yesterday that some dialogs, such as those for scenco don't load large enough to see all components. I'll look into this.

martinjlowm, 2016-05-26:
It seems to me (with workcraft-dist-template in mind) supplying a fully packaged application including the compiled binaries will be the correct approach (similar to how You distribute it for Windows and Linux at the moment). I have made a Cask formula [workcraft.rb](https://github.com/martinjlowm/homebrew-cask/blob/workcraft_cask/Casks/workcraft.rb) as I envision it to be when released. Let me know what packaging format you choose (the formula assumes .dmg at the moment). Once You release 3.1.0 I will try to have it accepted upstream in the official Cask repository.

Are previous versions removed from the website, when a new one is released? If so, I would advice You to wait removing the older version until the Cask formula is updated accordingly.

danilovesky, 2016-05-26:
@martinjlowm: Here is our first attempt to support OSX:
https://drive.google.com/file/d/0B2s4C-vvCyaiR2lieHoxczBHLVU/view?usp=sharing

Could you give it a try please (STG and Circuits are of top priority). Does synthesis and verification with Petrify and MPSat work for you? 

For distribution I was thinking about a basic tar.gz file. It would be great to do a proper DMG file, but the developers are mostly Linux users and we are not sure how to do it from Linux. Do you have experience?

Yes, we can keep the previous versions at http://workcraft.org/. They are also available  at https://github.com/tuura/workcraft/releases . Is it possible to add github as a mirror to the Cask formula?

martinjlowm, 2016-05-26:
> @martinjlowm: Here is our first attempt to support OSX:
> https://drive.google.com/file/d/0B2s4C-vvCyaiR2lieHoxczBHLVU/view?usp=sharing

Cool!

> Could you give it a try please (STG and Circuits are of top priority). Does synthesis and verification with Petrify and MPSat work for you?

I'll give it a try and get back to answering the question.

> For distribution I was thinking about a basic tar.gz file. It would be great to do a proper DMG file, but the developers are mostly Linux users and we are not sure how to do it from Linux. Do you have experience?

Not really, no. I've recently started using OS X and I was previously a Linux user as well. But I can certainly figure something out!

> Yes, we can keep the previous versions at http://workcraft.org/. They are also available at https://github.com/tuura/workcraft/releases . Is it possible to add github as a mirror to the Cask formula?

I see. Unfortunately no, you can read the reasoning here: https://github.com/caskroom/homebrew-cask/issues/10241

martinjlowm, 2016-05-26:
> Could you give it a try please (STG and Circuits are of top priority). Does synthesis and verification with Petrify and MPSat work for you?

Running any of the MPSat tools results in a linking error:
```
dyld: Symbol not found: __ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev
Referenced from: /Users/martinjlowm/Downloads/woooorky/workcraft/tools/UnfoldingTools/punf
Expected in: /usr/lib/libstdc++.6.dylib
 in /Users/martinjlowm/Downloads/woooorky/workcraft/tools/UnfoldingTools/punf
```
on OS X 10.11.4.

All Petrify synthesis works, except for the technology mapping. No real errors occur in stderr, but the error dialog is shown and I believe the process exit signal triggers it. The technology mapping behaves similar to the Linux version, but it silently terminates when the mapping is supposed to happen. Petrify just outputs:
```
[STDERR] Opening library file '/Users/martinjlowm/Downloads/woooorky/workcraft/libraries/workcraft.lib'.
[STDERR] The STG has CSC.
```

Note the output is sent to stderr (should be output to stdout as info?). Also; all STGs I test on are claimed to have CSCs by Petrify, even after resolving the conflicts whereas MPSat finds no conflicts. These are perhaps unrelated bugs.

Apart from this it seems to be functional and I haven't found any other bugs yet.

danilovesky, 2016-05-26:
Looks like dynamic linking problem :( The binaries were compiled and tested in OSX 10.11.5. I then tried the binaries on my old macbook with OSX 10.10 and saw the same errors you reported. We thought that newer OSX 10.11.* should have the same set of libraries. Apparently not...

Do you have any idea how to statically link the binaries so there is no this dependency problem? It looks like there is no easy way with GCC: https://stackoverflow.com/questions/5259249/creating-static-mac-os-x-c-build

About CSC - `The STG has CSC.` means it has Complete State Coding (good thing), i.e. there is *no CSC conflict* (bad thing).

danilovesky, 2016-05-26:
@martinjlowm we have an idea how to statically link at least with `libstdc++` and `libgcc_s`. I will prepare another internal release tonight or tomorrow for you to test.

martinjlowm, 2016-05-26:
> @martinjlowm we have an idea how to statically link at least with libstdc++ and libgcc_s. I will prepare another internal release tonight or tomorrow for you to test.

Okay good. Just let me know.

> About CSC - The STG has CSC. means it has Complete State Coding (good thing), i.e. there is no CSC conflict (bad thing).

Oh yea... obviously. Never mind then! :)

danilovesky, 2016-05-27:
@martinjlowm: Here is a new build with statically linked backend tools. 
https://drive.google.com/file/d/0B2s4C-vvCyaiLTd2cVM5UFF4VDQ/view?usp=sharing

STG synthesis and verification works most of the time even on my old Macbook (OSX v10.10). The only problem so far is that Petrify segfaults on technology mapping (on any example, even on the most basic ones). All other synthesis mods and MPSat seem to work as expected.

We also tried to address some of High-DPI issues in this build. If you notice visually strange dialogs or buttons - please let us know.

danilovesky, 2016-05-27:
It looks like we need to roll back to old Petrify (v4.2) for the Mac distribution... The Intel binary on Petrify's homepage seems to work well even in technology mapping (https://www.cs.upc.edu/~jordicf/petrify/distrib/petrify-4.2-mac.tgz)

danilovesky, 2016-05-27:
The progress is good for the first iteration. All the backend tools are working fine on OSX.

closed 2016-05-27 by danilovesky

martinjlowm, 2016-05-27:
@danilovesky I will test it soon.

I have some changes for workcraft-dist-template and dist.sh for a real OS X application feeling for the second iteration. These changes consist of an ICNS file (based on the SVG icon), the distribution bundled as an OS X app and one step towards using OS X's global menu.

I will create a pull request when it's cleaned it up and I feel it's ready.

danilovesky, 2016-05-27:
Thanks @martinjlowm! 

Here is another internal release with Petrify v4.2 and few cosmetic changes:
https://drive.google.com/file/d/0B2s4C-vvCyaiTDU4cUVTaVVhZU0/view?usp=sharing

I am planing to release in the next couple of days (because of a Workcraft tutorial next Wednesday). It would be great if you could do a PR before that... No rush though :)

danilovesky, 2016-05-29:
@martinjlowm Workcraft v3.1.0 is released! Thanks for making it look more *native* on OS X. Please proceed with the Cask formulae.

martinjlowm, 2016-05-30:
@danilovesky: You are welcome :)!

Pull request is pending: https://github.com/caskroom/homebrew-cask/pull/21544