workcraft/workcraft #429: Lower CPD token count to 500

mvdan, 2016-03-21:
These are the current blockers:

```
Found a 112 line (785 tokens) duplication in the following files: 
Starting at line 312 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/BufferRenderer.java
Starting at line 309 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/GateRenderer.java

Found a 105 line (729 tokens) duplication in the following files: 
Starting at line 206 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/BufferRenderer.java
Starting at line 215 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/GateRenderer.java

Found a 91 line (574 tokens) duplication in the following files: 
Starting at line 114 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/BufferRenderer.java
Starting at line 127 of CircuitPlugin/src/org/workcraft/plugins/circuit/renderers/GateRenderer.java

Found a 233 line (1333 tokens) duplication in the following files: 
Starting at line 148 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerAnalysis.java
Starting at line 127 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerTool.java

Found a 198 line (1135 tokens) duplication in the following files: 
Starting at line 188 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerAnalysis.java
Starting at line 225 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerQuery.java

Found a 196 line (1117 tokens) duplication in the following files: 
Starting at line 225 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerQuery.java
Starting at line 167 of XmasPlugin/src/org/workcraft/plugins/xmas/tools/VerTool.java
```

You can run `gradle cpdCheck` to get the full results, which include the lines in question.

mvdan, 2016-03-22:
Why make this blocking for 3.0.9? I would in general not make low-priority bugs blockers for releases.

danilovesky, 2016-03-22:
ok either way for me.

danilovesky, 2020-04-29:
CPD token count has been lowered to 250 with a couple of exceptions now.

closed 2020-04-29 by danilovesky