workcraft/workcraft #436: Workcraft throws an error when attempting to import a CPOG from a file after performing concurrency extraction

jrbeaumont, 2016-03-26:
This is using Workcraft version 3.0.9.

My operating system is Mac OSX, I am not currently near a Windows or Linux system to attempt to reproduce this on them.

To reproduce: 
Open a new CPOG file, and using the Process Mining menu, select "Import an event log".
Select this file from wherever you saved it:
[L2.txt](https://github.com/tuura/workcraft/files/190427/L2.txt)
Ensure to tick the "Perform concurrency extraction" option.
Click "Import"

I expected a CPOG to be inserted correctly. If you insert this same file from the CPOG algebra insert tool, using "Text File" from the tools section, it inserts correctly.

Instead, I see an error which suggests there is no graph name before the `=` and after dismissing this error, it has combined the last vertex from the first graph on the first line, and the name of the second graph on the second line. 

I will make some changes and possibly attempt to import it using the CPOG algebra plugin, specifically the inserting from a file.  

If someone could confirm for me whether this occurs on Windows or Linux, then I know whether it is a OSX specific issue and I can wait to commit anything until I know a fix of OSX does not affect Linux or Windows.

allegroCoder, 2016-03-26:
I followed your instructions on Linux. Below the error which pops up:
```
Lexical error
Lexical error at line 1, column 28. Encountered: "=" (61), after : ""
```

Hope this helps you out.

jrbeaumont, 2016-03-26:
Thanks mate.

This shows that it's at least affecting two operating systems, so a fix I make will help both Linux and OSX

closed 2016-03-27 by danilovesky