workcraft/workcraft #469: 1. 3.9

danhng, 2016-04-28:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?

2. What operating system and processor architecture are you using?

3. What did you do? If possible, provide a list of steps to reproduce
   the error.

4. What did you expect to see?

5. What did you see instead?

mvdan, 2016-04-28:
Wrapped the trace with triple backquotes to help readability.

danilovesky, 2016-04-30:
Thanks for the bug report! 
We just identified and fixed the cause of this problem few days ago -- see #466.
If you encounter this problem often and it blocks your work I can prepare an internal release for you.
Alternatively you can compile Workcraft from source.

closed 2016-04-30 by danilovesky