workcraft/workcraft #471: Use ~/.workcraft as fallback only

mvdan, 2016-05-01:
We should try, one after another:

* `$XDG_CONFIG_HOME/workcraft/`
* `$HOME/.config/workcraft/`
* `$HOME/.workcraft/`

Right now we jump directly to the third, at least on linux. The first two will often evaluate to the same as `$XDG_CONFIG_HOME` defaults to `$HOME/.config`.

Almost every up-to-date program uses `~/.config/` nowadays, and for good reason. We should too.

danilovesky, 2016-05-01:
What would be the preferred  path for Windows and OSX?

mvdan, 2016-05-01:
OSX: `$HOME/Library/Application Support/`
Windows: `%APPDATA`

danilovesky, 2016-05-01:
For `$HOME` there is a standard `user.home` property in Java API -- `System.getProperty("user.home");`. That is what we are using now.

I am not sure how to deal with `$XDG_CONFIG_HOME` and `%APPDATA%` properly though. We can use `System.getenv("XDG_CONFIG_HOME")` and `System.getenv("APPDATA")` but I am not sure it is a correct way.

mvdan, 2016-05-01:
It definitely is the right way for linux. If `$XDG_CONFIG_HOME` is not set, then its default value is `.config/`, as I explained above.

Actually, the third option can be skipped, as we can create `$HOME/.config/` if it doesn't exist yet.

PR #485 (2016-05-12): Standard OS-specific configuration directory
Closes #471.
commit 96be0512 (2016-05-12): Standard OS-specific configuration directory

Closes #471.
closed 2016-05-12 by danilovesky