workcraft/workcraft #472: None UTF-8 character in Custom Properties [MPSat] preventing load

DannyGardner, 2016-05-02:
Version 3.0.9
OS: Windows 10 64bit

Whilst writing a Custom Property, STG Reachability Analysis, I entered a non UTF-8 character "¬" without realising. I updated the property, and then ran the property. I received an error message when running, similar to the standard error you receive when entering in correct syntax of the Reach language.

After this you're unable to open the Custom Property panel, receiving the following error message in the Problems output:

```
Exception in thread "AWT-EventQueue-0" java.lang.RuntimeException: com.sun.org.apache.xerces.internal.impl.io.MalformedByteSequenceException: Invalid byte 1 of 1-byte UTF-8 sequence.
	at org.workcraft.plugins.shared.presets.PresetManager.<init>(PresetManager.java:56)
	at org.workcraft.plugins.mpsat.MpsatPresetManager.<init>(MpsatPresetManager.java:13)
	at org.workcraft.plugins.mpsat.tools.MpsatCustomPropertyChecker.run(MpsatCustomPropertyChecker.java:40)
	at org.workcraft.util.Tools.run(Tools.java:22)
```

When you go into the xml file where the Custom Properties are store and remove the character this fixes the problem and you're able to open the Custom Property panel again. 

It's also worth mentioning if you don't update the property and just run it with the character in the problem doesn't happen, only when properties are saved.

danilovesky, 2016-05-04:
I could not reproduce it in Linux (Ubuntu 16.04) -- the error message is very detailed (incorrect character in a specific position). The *Custom property* editor can be opened and the the character can be removed with no issues.

Now I need to find a Windows computer to see why it is a problem there...

DannyGardner, 2016-05-04:
Let me know if I can provide any more details or help, I'd be more than happy to.

vkhomenko, 2016-05-04:
We had a similar issue last year. The reason was that when storing this property in an XML file, a non-XML character was inserted. Then, when trying to open that dialog again, the parser fails to parse the file.

danilovesky, 2016-05-04:
Yes, I remember fixing that problem -- now I can save/load UTF8 and UTF16 characters to/from presets, at least in Linux.

commit 031e4736 (2016-05-10): Forbid non-ASCII characters in Reach expressions

Fixes #472.
closed 2016-05-10 by danilovesky