workcraft/workcraft #483: Run pgminer as a Workcraft task

snowleopard, 2016-05-12:
Please answer these questions before submitting your issue. Thanks!

> What version of Workcraft are you using?

3.0.9

> What operating system and processor architecture are you using?

Windows, x64

> What did you do? If possible, provide a list of steps to reproduce
   the error.

* Create CPOG model
* Process mining -> Import an event log
* Check "Perform concurrency extraction" and click Import

> What did you expect to see?

Process mining tool `pgminer` appearing in Tasks, Workcraft still responsive to other commands.

> What did you see instead?

Workcraft unresponsive to other commands, appears to be waiting for `pgminer` to finish.

@jrbeaumont I guess you are the best person to try to tackle this.

jrbeaumont, 2016-05-12:
I'll look into this now. Could you upload a file for me to test it with?

snowleopard, 2016-05-12:
I used `DigitalCopier.tr` from our benchmarks -- you should have it.

jrbeaumont, 2016-05-12:
I found `DigitalCopier.tr` and did a test, and `pgminer` runs fine. The issue comes when it is trying to insert the result into Workcraft, after `pgminer` has finished. We have had this issue before, and it is in the way Workcraft deals with CPOGs on such a large scale.

snowleopard, 2016-05-12:
I see. I think it may be useful to have a way to run certain Java-land code inside Tasks too. Otherwise we'll be hitting the same issue in other plugins with non-trivial code, which is not delegated to backend tools.

jrbeaumont, 2016-05-12:
It is possible, and I can arrange this, but it may be for either when any CPOG is inserted, or some code can be copied and it can be for any result from `pgminer`. Do you have any preference?

snowleopard, 2016-05-12:
On a second thought I don't think this will be easy to implement in this case. Fundamentally I think the problem may be not with the process mining plugin, but that Workcraft is too slow to deal with so many scenarios (inserting them in this case). We may consider this issue as a good benchmark for optimising Workcraft.

However, if you find that the bottleneck is actually your piece of code somewhere then there are two options: 1) optimise the code to make it faster, or 2) wrap that particular piece of code into a Task.

danilovesky, 2016-05-12:
If (a properly implemented) insertion loop takes so long then most likely the model would not be usable in Workcraft anyway.

snowleopard, 2016-05-12:
@danilovesky Unless we find a way to make Workcraft faster!

jrbeaumont, 2016-05-12:
I will have a look for more bottle necks. I remember fixing several last time we had this issue, but I am sure there is always room for improvement.

danilovesky, 2016-05-12:
@jrbeaumont: You may want to use a profiler to find the bottlenecks. I found VisualVM quite easy to use, but there are few other tools (https://blog.idrsolutions.com/2014/06/java-performance-tuning-tools/).