workcraft/workcraft #537: Improvements for OS X and high DPI displays.

jrbeaumont, 2016-06-22:
- [ ] Set `CMD` as default menu shortcut key on OS X
- [ ] Include option for selecting between menu shortcut as  `CMD` and `CTRL` on OS X
- [ ] Resize dialogs using `sizeHelper` so they all fit normally on both normal and high DPI displays
- [ ] Show prompt for `CTRL + RMB` to move the editor when `CTRL` held down
- [ ] Set `backspace` to work to delete nodes in the editor
- [ ] Use native OS X menu bar for Workcraft menus.
- [ ] Ensure none of these changes negatively affect Linux/Windows.

jrbeaumont, 2016-06-22:
I have done some tests with `meta mask` and `defaultMenuKeyMask()` on all three platforms. 

`meta mask` seems to not work on any platform. For Windows it should be the windows key, and `CMD` on OS X, I think it should be the absent windows key/cmd on Linux. None of these appear to work. 

`defaultMenuKeyMask()` uses `CTRL` on Windows and Linux, and `CMD` on OS X, and all work as expected.

What I suggest is that we use `defaultMenuKeyMask()` which will keep the normal `CTRL` key on Windows and Linux, and the standard `CMD` for OS X. We can then add an option for OS X users to select between the native `CMD` key, or using `CTRL` which has been used up to now.

danilovesky, 2016-06-22:
I think we use `Ctrl` and `Shift` most of the time. I only can remember `Alt` is used for page grouping. Is there any other use of `Alt`?

It may be a good idea to have `getCtrlMask()` static method in some globally used class that would take care of those user settings and return the correct mask for each platform. I see bait of confusion with `Cmd` as all the help and tutorial files refer to `Ctrl` key. We may need to adjust the help text accordingly.

jrbeaumont, 2016-06-22:
I cannot find any other than `Alt+F4` to close. I beleive we can leave `Alt` and `Shift`, there should be no change there.

That's a good idea. I am aiming to have the setting hidden for Windows and Linux and it only return `Ctrl` (`CTRL_MASK`), but for OS X it either returns `Ctrl` (`CTRL_MASK`) or `Cmd` (`defaultMenuKeyMask()`) depending on their choice.

We could add a note in tutorial saying that every use of `Ctrl` can be replaced by CMD for OS X.

jrbeaumont, 2016-06-22:
I may have to make a similar method for the purpose of getting the name of the key used, as some tooltips refer to `Ctrl`.

jrbeaumont, 2016-06-22:
Quick update: `meta_mask` appears to work for OS X now, which will make things a little easier. 

No idea what I was doing before, but it's now simpler at least.

jrbeaumont, 2016-06-24:
@danilovesky I have found an issue with panning. 

By default `Ctrl+directon-key` is a command in OS X to move between desktop screens. This means that it doesn't work when `Ctrl` is selected as the menu shortcut in Workcraft on OS X. Shall I set it to be `cmd` by default only on Macs?

danilovesky, 2016-06-24:
@jrbeaumont I think we should switch to `Cmd` completely as soon as it works smoothly. Maybe even without an option to use `Ctrl`...  I think we should have a function `getMenuShotcut()` in `DesktopApi` that returns `Ctrl` or `Cmd` depending on the OS.

If you feel it is ok to leave OSX users without a Ctrl/Cmd choice -- you can even do it even now. A lot of your recently added code will get redundant though :(

jrbeaumont, 2016-06-24:
My thoughts were by giving the option for users who may have used it on OS X before, where `Ctrl` has been used by default before now, may want to go back to how they're used to. Switching it to `cmd` by default would make this code redundant, but that's not a problem :) 

Which would you prefer? I can do either, no problem :)

jrbeaumont, 2016-06-26:
Just while we make a decision on this, I have noticed that text boxes still use `ctrl` by default for things like select all and copy/cut and paste. I'll try and fix this.

In the mean time I have found and implemented a way to use both `backspace` and `delete` for deleting selected elements.

jrbeaumont, 2016-06-27:
My repo currently implements `CMD` as default for shortcuts etc in OS X, which has not affected it on Windows, Linux is yet to be tested, but I assume it will work.  `backspace` works as well as `delete` to delete nodes on all platforms. 

So far, for using `cmd` for shortcuts in text boxes, there is a fairly large issue. Mainly, there is no easy way to just ask them to replace `Ctrl` for `CMD` in the existing shortcuts. Nor is it apparently simple to remove existing shortcuts, and replace them with ones that perform the same operation but using `CMD`, at least not without removing all keybindings, which will mean you can't actually type in it without a process of replacing them all. 

In this case, I suggest to simply just add `CMD` shortcuts on all text boxes on OS X, which will mean they should work in addition to the normal `Ctrl` shortcuts. Not a major issue, but at least there is the option.

danilovesky, 2016-06-28:
@jrbeaumont I think the text box shortcuts can be set globally (in SilverOceanTheme class) as explained here:
http://stackoverflow.com/questions/7252749/how-to-use-command-c-command-v-shortcut-in-mac-to-copy-paste-text

jrbeaumont, 2016-06-28:
@danilovesky So you can, I'll give this a try.  
Thanks!

danilovesky, 2016-06-28:
It may be better if I add this change (permanently) to SilverOceanTheme as I am significantly changing that class at the moment. @jrbeaumont could you try if it works on mac (as a temporary change)?

jrbeaumont, 2016-06-28:
Yeah I will. Do you have any preference about where it goes in the class, so I can test it a bit closer to (future) reality?

danilovesky, 2016-06-28:
Just put that code in the end of `enable()` method - it should do.

jrbeaumont, 2016-06-28:
This doesn't appear to be working. I'm looking into it however.

danilovesky, 2016-06-28:
Try moving the code to `addCustomEntriesToTable`...

jrbeaumont, 2016-06-28:
So it appears that the `InputMap` you get from `UIManager.get("TextField.focusInputMap")` does already include the `meta` key in OS X, including things like Cut/Copy/Paste, but this isn't automatically added to any form of TextComponent.

jrbeaumont, 2016-06-28:
I have done some tests, and it appears that the `InputMap` you get when in classes like `CpogSelectionTool` is different than the one you get when in `SilverOceanTheme`, so when setting up the theme, it aims to use `CMD`, but somewhere along the way it changes to using `CTRL` etc again.

jrbeaumont, 2016-06-28:
@danilovesky Found it. 

In `LookAndFeelHelper`, when line 59, `UIManager.setLookAndFeel(laf);` is called, this set's the standard key commands to use `Ctrl` etc.

danilovesky, 2016-06-28:
@jrbeaumont What if you do it after `LookAndFeelHelper.setDefaultLookAndFeel();` in `MainWindow`?

jrbeaumont, 2016-06-28:
I have been testing it inside `LookAndFeelHelper.setDefaultLookAndFeel()`, and I have to reset the `InputMap` for both `TextField.focusInputMap` and `TextArea.focusInputMap` for larger text boxes, such as in the algebra plugin for CPOGs, to use `CMD` for shortcuts. This works great!

jrbeaumont, 2016-06-29:
@danilovesky I have just added the functionality so the OS X menu bar is now used. 
However, I have had to include a mac specific class, `AquaMenuBarUI`, as this is what specifies whether it stays in the `JFrame` or can be put into the native menu bar.

This class is included with Oracle java for OS X only as far as I can tell.

As I expected, this is causing issues when assembling, as obviously `AquaMenuBarUI` isn't standard. And looking at it build on travis (https://travis-ci.org/jrbeaumont/workcraft) you can what it's issues are.

I sensed this might be an issue, so is there a way we can include it? Or make it platform specific? Or shall we just remove it and not bother?

jrbeaumont, 2016-06-29:
Never mind, I have found a way to avoid using specific classes, and builds correctly with gradle.

I'll test it on Windows tonight.

closed 2016-06-30 by danilovesky