workcraft/workcraft #560: Integrate concepts tool into STG plugin

jrbeaumont, 2016-08-02:
- [ ] Add `Concepts` menu in STG plugin.
- [ ] Implement importing of concepts from a `.hs` file.
- [ ] Implement authoring of smaller test concepts in dialog for translation to STG.
- [ ] Add helpful error messages in case of GHC not being installed/concepts being wrong/concepts not installed.

danilovesky, 2016-08-03:
Just to clarify, do we go for a dedicated `Concepts` menu or would it be sufficient to have it as one of the importers from `.hs` file (similar to import from `.g` and `.sg` files)?

jrbeaumont, 2016-08-03:
Personally I would prefer a `Concepts` menu, for both importing from a `.hs` file, and writing small concepts in a dialog for quick tests in the STG plugin. 

I was also aiming to set it as an importer for the `File > Import` menu. 

Is this OK?

If you would prefer, we could put this in the `Conversion` menu which already exists in the STG plugin as that is effectively what we're doing. This would save menu bar space.

danilovesky, 2016-08-04:
I think this should be implemented as an import plugin (for `Import` menu).  Having it in under `Concepts` menu may also make sense, especially if there is more than one menu item related to concepts. We can experiment with the menus/submenus later.

jrbeaumont, 2016-08-04:
I can do that. 
Can I put this within the `STGplugin`, or would it be best having a separate `ConceptsPlugin`?

danilovesky, 2016-08-04:
I would put the importer in `org.workcraft.plugin.stg.interop`, similar to `DotGImporter` class and create a package `org.workcraft.plugin.stg.concepts` for all the parsing and backend calling stuff.

PR #563 (2016-08-09): Concepts can now be written, translated and inserted to Workcraft.
This has been tested on OS X and Windows, on both High-DPI and 1080p displays, and the dialog looks correct, and the functionality also works correctly.

This should close #560
closed 2016-08-09 by danilovesky