workcraft/workcraft #561: OS X cannot run external installed tools

jrbeaumont, 2016-08-02:
Installed command line tools such as `dot` cannot be run through Workcraft in OS X.

This applies to the Workcraft.app installation, downloaded from the website.

The error given is:
```Cannot run program "dot" (in directory "[temp-file-directory]"): 
error=2, No such file or directory```

This seems to be an issue of Workcraft not possessing the system `PATH`.

jrbeaumont, 2016-08-02:
Some testing reveals that this only occurs when you run Workcraft by double-clicking the icon.

Running the workcraft start-up script from `Workcraft.app/Contents/MacOS` allows `dot` to be run correctly.

jrbeaumont, 2016-08-02:
A workaround:

Add `export PATH="/usr/bin:/opt/local/bin"` to the Workcraft start up script. 

This may not be ideal, as these three are needed specifically for `java` and `dot`.

jrbeaumont, 2016-08-02:
Yet more information revealed from testing.

The default `PATH` just for `Workcraft.app` is "/usr/bin", which is the default place for the `java` executable. This is also the case for every `.app`in OS X, it doesn't use the `PATH` defined in  the terminal, unless you start an app from the terminal. Go Apple.

Thus, the start up script can include `export PATH="$PATH:/opt/local/bin"`, which is where I have `dot` installed. 

This whole thing started because I realised that I couldn't run `runghc` for testing the concepts tool as per #560. So I have also been including `/usr/local/bin`, for where this is installed.

jrbeaumont, 2016-08-02:
Installing `graphviz` for OS X directly from the website installs it in `/usr/local/bin`, meaning we could probably just include the one extra line in the script:
`export PATH="$PATH:/usr/local/bin"`
And suggest to users that if `dot` doesn't work, they install it from their website directly.

`runghc` will be an issue for now, but this is only used as there are issues with the binary for the concepts tool, which shouldn't be a permanent issue.

danilovesky, 2016-08-03:
Is there a native Graphviz distribution for OSX or did you install it via macports? If installing Graphviz is not straightforward, then we can include a cut-down version (only dot, similar to what we did for Windows).
Another option is to call dot (and maybe ghc) by the absolute pat -- check if `/usr/local/bin/dot` is present first).

jrbeaumont, 2016-08-03:
There is a native one you can download on their website and install easily, download, double click, follow instructions. This binary is available from `/usr/local/bin`. 
I initially had a Mac ports version installed which was the issue, and this version was available from `/opt/local/bin/dot`.
The issue here is that a user could install via either Mac ports (or home brew) or from a website just as easily, for both Haskell and Graphviz, and if we use the absolute path to `/usr/bin/local/` for either dot or runghc and a user has the Mac ports version installed Workcraft can't use it.

danilovesky, 2016-08-04:
It looks like Finder (of whatever that GUI thing for file browsing is called), has its own settings for PATH (defined in `/etc/paths`) and does not respect `~/.bash_profile` settings.
The easiest option may be to add several most used path to workcraft script, .e.g:
`export PATH=$PATH:/usr/local/bin:/opt/local/bin`
If the some of the paths do not exist -- it should not be a problem.

jrbeaumont, 2016-08-04:
I will add those then and send a PR for workcraft-dist-template.

danilovesky, 2016-08-04:
Test it locally a bit longer -- there is no release till September :)

closed 2016-09-19 by danilovesky