workcraft/workcraft #603: Upscaling on HiDPI displays causes pixelated text

mvdan, 2016-11-05:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?

`master` as of writing, 926790564844209c14889c0e88569bbed08510d2.

2. What operating system are you using?

Linux 64-bit.

```
$ java -version
openjdk version "1.8.0_112"
OpenJDK Runtime Environment (build 1.8.0_112-b15)
OpenJDK 64-Bit Server VM (build 25.112-b15, mixed mode)
```

3. What did you do? If possible, provide a list of steps to reproduce the error.

Start workcraft on my `2560x1440` laptop.

4. What did you expect to see?

Upscaling of buttons and text.

5. What did you see instead?

The upscaling, but with pixelated text. Attaching screenshot of some of the buttons with text.

The same happens with icons, but text is the most obvious and annoying one. It should also be the easiest to fix since I don't know if icons are available in vector format, whereas text can be rendered at any size.

[image]

CC @jrbeaumont

danilovesky, 2016-11-05:
@mvdan did you see this issue before? It seems to render ok for high-resolution on Windows and OSX under Oracle JRE 1.8 (not sure about Open JDK). I do not have a high-resolution Linux machine though.

@snowleopard did you notice text pixelation on your new laptop?

snowleopard, 2016-11-05:
@danilovesky No, I don't have this issue on my laptop.

However, in general, there are a lot of problems due to high-resolution. We'll need to sit down at some point and discuss what would be the best way to address most of them without much work.

jrbeaumont, 2016-11-06:
I find that openJDK on a virtual machine running ubuntu, the text is quite pixelated also.

jrbeaumont, 2016-11-06:
Maybe disregard that, it's pixelated with Oracle JDK too, so it's likely just a bad VM

mvdan, 2016-11-06:
> However, in general, there are a lot of problems due to high-resolution. We'll need to sit down at some point and discuss what would be the best way to address most of them without much work.

To clarify - the current state of things is much better than when no upscaling was done. At least I can see the buttons, read the text and click on stuff. This is simply a bug report for something that isn't working quite right, but it's definitely not a call for a rethink of the whole HiDPI support.

@danilovesky I'll try some older versions now to see if it's a new thing. I never noticed any pixelation, but I probably wasn't paying enough attention anyway.

In any case, if other platforms or JDKs do it fine, then it's most likely due to the combination of OpenJDK and Linux and not a bug in workcraft.

mvdan, 2016-11-06:
It happens on 3.1.0 too, so it's been around ever since HiDPI support was introduced.