workcraft/workcraft #605: Right-click (two-finger click) doesn't work on Mac trackpad.

jrbeaumont, 2016-11-08:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?
3.1.2
master from tuura/workcraft

2. What operating system are you using?
macOS 12 Sierra
I am unsure whether this is because I updated my mac.
I have looked into how right-click works on macs, and it appears that right-click works in the same way for all macs using the trackpad.

3. What did you do? If possible, provide a list of steps to reproduce
   the error.
Tried to right-click a gate in the Digital Circuit plugin in order to add a new input.

4. What did you expect to see?
A popup menu

5. What did you see instead?
Nothing

jrbeaumont, 2016-11-08:
I will do some testing for a few things:
- Does a USB mouse work with right-click
- Are there any settings that can change how a right-click works with a trackpad.
- How it manages to ignore a right-click.

danilovesky, 2016-11-08:
I also noticed this issue when helping a student on his macbook.
On my old macbook pro updated to 10.10.5 two-finger right-click also does nothing even though I *think* it used to work few months ago. Mouse right button is also ignored :(

These seem to be useful threads:
http://stackoverflow.com/questions/2972512/how-to-detect-right-click-event-for-mac-os
http://stackoverflow.com/questions/4525733/java-mouse-event-right-click

jrbeaumont, 2016-11-08:
No settings to change how the track-pad works with right click :(
I think I may know what's happening. Let me just test it.

danilovesky, 2016-11-08:
Looks like we need to use `SwingUtilities` API instead of testing the events explicitly.

jrbeaumont, 2016-11-08:
I think what's happening is when you right-click on a mac, it sends it through to java as mouse button 3, but with the modifier of the CMD key. This means that when a mouse click is registered, it's ignored by `GraphEditorPanelMouseListener` because it will be the `panKeyCombo`. 
This also reveals that you can pan the editor by just holding down right-click (two fingers) on mac.

danilovesky, 2016-11-08:
Can you check if `SwingUtilities.isRightMouseButton(MouseEvent anEvent)` helps to register right clicks? Most likely this wrapper function handles right-click events differently for OSX, Linux and Windows.

jrbeaumont, 2016-11-08:
It does, but this still registers the meta key being pressed, so `isPanCombo()` still returns true.

jrbeaumont, 2016-11-08:
From `mouseClicked(MouseEvent e)` I removed 
`if (!isPanCombo(e))`
and the end to this if statement, and it seems to register right clicks correctly.
And pressing, CMD and holding right click allows me to pan.

jrbeaumont, 2016-11-08:
Removing this if statement still allows right-click and panning with Ctrl+right-click to work on Windows.

danilovesky, 2016-11-08:
Do you think just removing the if-statement would solve the problem? Did you check if it still works in Linux?

jrbeaumont, 2016-11-08:
I haven't checked Linux yet. 
I do think it would work, providing it still works in Linux.
A mouseClicked event isn't triggered when you try to do a click and drag.

jrbeaumont, 2016-11-08:
@danilovesky 
Linux works as well.
I will send a PR

PR #606 (2016-11-08): Right-click (two-finger-click) should now work on macOS
Fixed #605
commit 34a928e2 (2016-11-08): Right-click (two-finger-click) should now work on macOS

Fixed #605
closed 2016-11-08 by danilovesky