workcraft/workcraft #645: Opening a new work opens two new works (2 tabs) of the same type.

jrbeaumont, 2017-01-16:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?
    Workcraft master repo
2. What operating system are you using?
    Occurs on Windows and macOS
3. What did you do? If possible, provide a list of steps to reproduce
   the error.
    Run Workcraft, open a new work via CTRL+N (CMD+N for macOS) and select anything. 
4. What did you expect to see?
    Open only one new work.
5. What did you see instead?
    Opened two new works.

danilovesky, 2017-01-16:
Is it reproducable or happens only from time to time?

jrbeaumont, 2017-01-16:
It happens every time for me.

jrbeaumont, 2017-01-16:
If it helps, I have identified the code responsible.

`org.workcraft.Framework`, lines 572-574, this creates a new editor window.

`org.workcraft.gui.MainWindow`, lines 701-703, this also creates a new editor window, after having called the method in `Framework`. 

I am not sure which is preferred, or which was original, but these both do the same thing, based on whether the user has chosen to open the work in the editor (visually). 

I would search for other uses of both of these methods, to find out if one of these is preferred over the other, but gradle doesn't seem to want to work through eclipse at the moment, and I don't know why.

danilovesky, 2017-01-16:
@jrbeaumont thanks for debugging. I know the reason, need to decide what to do about it...

commit c2d41753 (2017-01-20): Simplified create work dialog

Only the model type can be selected, redundant checkboxes
and title text box are removed.

Fixes #645
closed 2017-01-20 by danilovesky