workcraft/workcraft #665: Refactor Plato into an external tool and add FST translation

jrbeaumont, 2017-03-07:
Plato has some new features which we aim to add functionality to Workcraft to use.

- [ ] Extract Plato files into an external `PlatoPlugin` directory

- [ ] Tidy plato code, to be more understandable, and so it doesn't need to close anything.

- [ ] Add translation from concepts to Finite State Transducers. 

- [ ] Run check for the invariant automatically on STG translation.

jrbeaumont, 2017-03-08:
@danilovesky
I have managed to extract the files successfully, gradle builds Workcraft sucesfully. I am now just aiming to set it up to work again, I'll let you know how this goes.

jrbeaumont, 2017-03-09:
@danilovesky: I have a question. 
I extend `AbstractConversionCommand` for Plato, but I do not use the `convert` method (i.e. It returns `null`). 

Should I use `convert`, or am I OK to switch from `extend AbstractConversionCommand` to `implements Command` and add this to the `Conversion` menu?

jrbeaumont, 2017-03-09:
My reasoning for turning it into a `command` instead is that it doesn't strictly implement a `AbstractConversionCommand` as it doesn't convert one model entry to another.

danilovesky, 2017-03-10:
There are two methods doing similar thing: `convert` and `run`. The former is blocking and returns the resultant model, the later is non-blocking and does not return anything explicitly. That `convert` command was introduced recently to better support scripting. Ideally only a `convert` method should be implemented and default implementation of `run` just calls `convert` asynchronously.

jrbeaumont, 2017-03-10:
OK, I will try and make `convert` work.

danilovesky, 2017-04-06:
Closed by #668

closed 2017-04-06 by danilovesky