workcraft/workcraft #667: Automated insertion of mutex element on STG synthesis

danilovesky, 2017-03-13:
1. Recognise mutex pattern in STG , based on a simple heuristic: conflict on two output signals with the same triggers for set and reset phases of these signals.
2. Extend STG with explicit mutex snippet, so the conflict is factored out into environment.
3. In the synthesised circuit insert a mutex element in place of the inserted helper signals.

vkhomenko, 2017-04-21:
Below is a REACH expression for checking the possibility of mutex insertion.

```
// For given signals r1, r2, g1, g2, check
// whether g1/g2 can be implemented by a
// mutex with requests r1/r2 and grants g1/g2.
// It is very similar to the "strict implementation",
// i.e. we are looking for a "bad" state where 
// the next-state values of the grants differ from
// [g1]=r1&~g2 and [g2]=r2&~g1

let
  r1=S"r1",
  r2=S"r2",
  g1=S"g1",
  g2=S"g2"
{
  ('g1 ^ ($r1 & ~$g2))
  |
  ('g2 ^ ($r2 & ~$g1))
}
```

vkhomenko, 2017-04-21:
Note that in the above REACH expression, WC must pass the correct names for the four signals in the `let` statement.

danilovesky, 2017-10-06:
Mutex place implementability check is currently too restrictive as it fails for "early release" version of the arbiter protocol where a grant may be issued while the other grant is still high (but after its request is withdrawn). (It corresponds to adding buffers at the grant outputs of mutex.) This protocol can still be implemented by a mutex.

@vkhomenko propose to modify the check as follows:

  1. Constraints on the next-state functions of the grants:
    `r1*~g2 => nxt(g1)`
    `~r1 => ~nxt(g1)`
    `r2*g2 => ~nxt(g1)`
   (and symmetric constraints for `nxt(g2)`).

The only state when the value of `nxt(g1)` is not implied by these constraints (i.e. is don't-care) is `r1*~r2*g2`, which is the flexibility of choosing between the early-release and late-release protocols.

Interestingly, these constraints don't imply that the critical sections are mutually exclusive! That is, adding the constraint `r1*g1*r2*g2` to the above constraints will not lead to a contradiction, and will simply imply `~nxt(g1)*~nxt(g2)`. It looks like these constraints will maintain the invariant `~(r1*g1*r2*g2)` once it's satisfied. However, I still think it's worth adding the following constraint to check the violation of mutual exclusion at the initial state:

  2. Mutual exclusion of critical sections:
`~(r1*g1*r2*g2)`

The corresponding REACH property is as follows (note that the above constraints had to be negated there to express the "badness" of a state):
```
// For given signals r1, r2, g1, g2, check whether g1/g2 can be implemented
//  by a mutex with requests r1/r2 and grants g1/g2.
// The properties to check are:
//   r1&~g2 => nxt(g1)
//   ~r1 => ~nxt(g1)
//   r2&g2 => ~nxt(g1)
// (and the symmetric constraints for nxt(g2)).
// Furthemore, the mutual exclusion of the critical sections is checked:
// ~( (r1&g1) & (r2&g2) )
// Note that the latter property does not follow from the above constraints
// for the next state functions of the grants.

let
  r1s = S"r1",
  g1s = S"g1",
  r2s = S"r2",
  g2s = S"g2",
  r1 = $r1s,
  g1 = $g1s,
  r2 = $r2s,
  g2 = $g2s,
  g1nxt = 'g1s,
  g2nxt = 'g2s
{
  // constraints on nxt(g1)
  r1 & ~g2 & ~g1nxt // negation of r1&~g2 => nxt(g1)
  |
  ~r1 & g1nxt // negation of ~r1 => ~nxt(g1)
  |
  r2 & g2 & g1nxt // negation of r2&g2 => ~nxt(g1)
  |
  // constraints on nxt(g2)
  r2 & ~g1 & ~g2nxt // negation of r2&~g1 => nxt(g2)
  |
  ~r2 & g2nxt // negation of ~r2 => ~nxt(g2)
  |
  r1 & g1 & g2nxt // negation of r1&g1 => ~nxt(g2)
  |
  // mutual exclusion of critical sections
  r1 & g1 & r2 & g2
}
```

danilovesky, 2017-11-03:
At the last step is missing: In the composed STG restore the original types of mutex grants.

PR #684 (2017-05-03): Synthesis and verification of STGs with arbitration
Closes #667 and #646
PR #735 (2017-10-07): Improved mutex place implementability check
Now "early release" version of the arbitration protocol are allowed.

Closes #667
commit 63be4220 (2017-10-07): Improved mutex place implementability check

Now "early release" version of the arbitration protocol are allowed.

Closes #667
PR #752 (2017-11-03): Fix verification of circuits with mutex
Restore the original types of mutex grants in the composed STG before
proceding to the verification.

Fixes #667
closed 2017-11-03 by danilovesky