workcraft/workcraft #674: Importing .sg files with internals throws an exception.

jrbeaumont, 2017-04-19:
Importing a .sg file which has any internal signals throws an error, some of which information is found here:
[sg-import-error.txt](https://github.com/workcraft/workcraft/files/933693/sg-import-error.txt)

It appears that it struggles with a label colour. 

This affects importing through Plato as well, but not when converting from something like an STG.

jrbeaumont, 2017-04-19:
Further testing reveals an issue with plato and the .sg files I was using. 

Internal signals in .sg files need to be referenced with `.internal`, but plato produces an .sg with `.internals`.

I will fix plato, and ask for the new version to be included. Then this can be closed

jrbeaumont, 2017-04-19:
Plato is now fixed, the latest version works correctly with Workcraft. 

Please update the version in the dist-template repo and this issue can be closed.

danilovesky, 2017-04-19:
Thanks, I will fix the parser of `.sg` too.

jrbeaumont, 2017-04-19:
In what way? I thought `.internal` was correct?

danilovesky, 2017-04-19:
yes `.internals` is incorrect but the parser should probably error out, or give a warning.
Currently all keywords starting with `.` that are not recognised are just ignored.
I will write a warning for such keywords, just in case..

jrbeaumont, 2017-04-19:
Ahh OK. I was confused. 

Thank you!

danilovesky, 2017-04-19:
Could you remind please why do you have this line in PlatoImporter?
```
if (output.startsWith(".model")) {
```

jrbeaumont, 2017-04-19:
In case there was an error in the translation. Instead of creating an actual file, we use `stdout` and if it is an error, it will not start with `.model`.

danilovesky, 2017-04-19:
I see.. should there be a warning if it does not start with `.model` or silent behaviour is ok?

jrbeaumont, 2017-04-19:
If it doesn't start with `.model` it throws a `PlatoException` which handles the error.

danilovesky, 2017-04-19:
Warning is issued on unsupported keywords -- fixed in #675.
Palto backend is updated to prevent incorrect `.internals` keyword.

closed 2017-04-19 by danilovesky