workcraft/workcraft #685: Improve detection of gС set and reset functions

danilovesky, 2017-05-04:
Suppose we have a gC implementation:
`x = expr`, where `expr=set + x* ~reset`, where `set`/`reset` do not depend on `x` and `~(set*reset)` is always true, i.e. `reset => ~set`.

Set and reset functions can be extracted as follows:
  * Set: `expr[x=0] = set`
  * Reset: `~expr[x=1] = ~(set + x* ~reset)[x=1] = ~set*(~x + reset)[x=1] = ~set*reset = [due to reset => ~set] = reset`.

danilovesky, 2017-08-24:
The `reset` expression obtained via `expr[x=1]` often appears to be too complex. To further simplify it one needs to apply `set=0` condition. This functionality is not implemented in our expression minimisation engine though (only literals can be assigned to a constant).

vkhomenko, 2017-08-24:
What do we need this for? I guess these set/reset functions are never shown for the user, but only needed for some internal purposes (e.g. to construct some Reach expression), so the complexity does not matter much.

danilovesky, 2017-08-24:
They are shown when the component rendering type is `Box` (default for gC implementation).

vkhomenko, 2017-08-24:
I guess we should show what the user had entered - if they entered set/reset then these are (hopefully) nice (at least the tool is not to be blamed). If they had entered the CG implementation, we should show it rather that set/reset.

A possible alternative is to implement SAT (e.g. by linking sat4j) and proper minimisation.

danilovesky, 2017-08-24:
gC gates are imported into set/reset functions, so it is our duty to make them "nice" :)
sat4j license may be problematic (needs more careful examination and discussion).
I leave the extraction of set/reset untouched for now...

danilovesky, 2020-10-09:
This issue has been addressed on MPSat and Petrify side via predefined formatting of the gC function as follows:
 `assign x = (SET_EXPR) + x * ~ (RESET_EXPR)`

The initially proposed solution  has been discarded as it could (significantly) spoil the structure of set/reset expressions.

closed 2020-10-09 by danilovesky