workcraft/workcraft #688: Automatization of circuit reset generation

sergeymileiko, 2017-05-08:
As a Workcraft user, I would like the process of circuit design to be mostly automated. Now, the circuit that I can generate from my STG specification is not completely finished. The circuit is not set to its initial state, and I need to analyze the circuit and add a reset signal to some logic gates.
Workcraft has an initialization analyzer tool to do this. It highlights, which gates will be set to their initial values after setting certain signals. However, this tool has the following disadvantages:

-  Everything is done by hand. You should reset the signals and analyze the gates by yourself.

-  You never know that the solution you have chosen is optimal.

-  You should draw the reset circuit by yourself.

The solution is to write a tool (or expand the initialization analyzer tool) that automates the process of reset generation. The tool will consist of the following modules:

- Input from user. The user can specify the inputs to his circuit, which initial state he definitely knows. Also, if the initial state does not matter, or if there is a set of allowed initial states, they can also be specified from user at the level of STG. Sometimes, it is much easier to reset the circuit to the particular initial state, than to some other states.

- Analyzer. The algorithm of this module can be the following:
Step 1: Set the input signals defined by user.
Step 2: Use the initialization analyzer to define the gates that have been set to their initial values. Exclude these gates from the further analysis.
Step 3: If there are no unset gates, finish the algorithm, otherwise go to the next step.
Step 4: Set each gate to its initial value, and define the Ni – the number of gates the set after setting the i gate. 
Step 5: Find the largest Ni and reset the I gate. Go back to the step 2.

- Reset generator. The module should generate the reset signal, and add it to the circuit according with the results of the analyzer. The circuit gates should be either expanded with additional reset input or some extra gates should be added (if a gate library restricts to expand the gate).

danilovesky, 2017-05-08:
@sergeymileiko: I am afraid there is no clear definition of *optimal solution* here. Your suggested algorithm (if I understood it correctly) may produce a solution that intends to modify the minimal number of gates. This may even translate to the smallest circuit size, not always though:

If a gate extended with reset is missing from the library, then you would need to insert an extra reset gate. A a smaller solution can sometimes be found by modifying two other gates whose "reset" version exist in the library. 

Moreover, often the circuit latency is much more important than its size. In this case other reset insertion strategies should be used.

sergeymileiko, 2017-05-09:
Yes, you are right, the issue will be more complicated. I've proposed just a straightforward solution, but more elegant ones definitely can be found.
Maybe, the correct solution will be to expand the existing gates only; and the algorithm can exclude the gates that cannot be expanded. Or just a certain combination of both approaches.

danilovesky, 2019-01-18:
Reset insertion is automated sufficiently well by _Initialisation analyser_ (https://workcraft.org/help/circuit/initialisation) and there is no easy way to fully automate this, so the issue can be closed.

closed 2019-01-18 by danilovesky