workcraft/workcraft #719: Resolve name conflict on import of hierarchical STG

danilovesky, 2017-07-03:
If an STG has hierarchical nodes (with ``.`` as hierarchy separator), then a conflict may appear in import in Workcraft, e.g.:
```
.inputs pg0.in
.outputs pg0.out
.graph
pg0 pg0.in
pg0.in pg0.out
pg0.out pg0
.marking {pg0}
.end
```
When reading such an input file Workcraft first creates a place ``pg0``, but then detects a page ``pg0`` with transitions ``in`` and ``out``. It causes a name conflict.

A possible solution is to preprocess the input file, conflicting names and create a map of renames for places (or maybe for pages) if necessary.

vkhomenko, 2020-05-12:
Renaming places is bad (e.g. can break custom properties referring to places). Best would be to have a separate name space for pages, or (internally) append '.' or some other character to page names.

danilovesky, 2020-05-12:
Agree, something safer needs to be done (that was the vision of the issue in 2017 and it needs to be revised).

danilovesky, 2020-05-15:
@vkhomenko after many frustrating hours I have to give up on changing how pages are represented and handledin Workcraft. Too many dependencies that I do not want to break (e.g. Circuit components, SON blocks, xMAS to STG conversion, WTG waveforms).

I have to roll back to place renaming on import of .g files. Note that backend tools can rename (or even remove) places anyway (e.g. as happens for mutex places on CSC resolution) and custom properties would not survive such transformation.

Also I think forbidding name clashes between pages and other types of nodes is correct design (it is similar to how namespace and variable names are handled in C++).

vkhomenko, 2020-05-15:
@danilovesky 

I think the argument about CSC resolution is invalid - note that Petrify re-synthesises the STG, i.e. the result may look very different from the original STG. I think we should accept that all synthesis modes (PN synthesis, CSC, logic decomposition) may completely change the STG.

Forbidding name clashes may well be a good idea - but you cannot enforce this, as the STG example at the top shows. WC probably already enforces this in the STG editor - the problem is the import of STGs from backends.

Let's not hurry - maybe we could think of a better solution?

vkhomenko, 2020-05-15:
@danilovesky
BTW, PCOMP currently renames places - should it rather encapsulate each component STG into a page and keep the original place names?

danilovesky, 2020-05-15:
@vkhomenko Please let me know if you have any ideas.

It looks like both Workcraft and backend tools work with the named "correctly". It is the interface between the two that needs to be "fixed".

I do not think pages would help reading PCOMP output. We already have details.xml to track renaming of places there. These still can clash with page names though :-(

commit b8813c4a (2020-06-08): Tests for STGs with page name conflicts

Fixes #719
closed 2020-06-08 by danilovesky