workcraft/workcraft #751: Verification of term overtaking in Circuit

vkhomenko, 2017-11-02:
We have a gate with output `O` computing function `F` that has a binate input `Bin`. This function  `F` is implemented as a _sum-of-products_ cover, perhaps with some redundant cubes.

Witness of property violation for the _positive phase_:
* `O` is enabled, and
* `Bin` is enabled, and
* `~dF/dBin`, and
* for each cube `C` in the cover: `~(C & ~dC/dBin)`.

A similar witness needs to be derived for the _negative phase_.

danilovesky, 2017-12-20:
How to enforce consensus covers in Petrify synthesis results?

### Option 1

For a binate signal `x` explicitly insert its inversion `x_bar` in a speed-independent way, so it is properly acknowledged. This insertion can be done mechanically at the level of a state graph (identify ERs and QRs for `x`, replicate the states, replicate arcs withing ERs and QRs, appropriately insert `x` and `x_bar` events between ERs and QRs -- see SIP insertion in the *brown* book for details).

**Main problems**:
This attempt has failed for D-latch example because there was no acknowledgement for the newly inserted signal. It still *may* work for other examples -- subject for research.

### Option 2

1. At the level of Boolean minimisation, when using Espresso:
    * The default way of Boolean minimisation is via BDDs. It is possible to use `espresso` for minimisation (`-boolmin4` option), maybe at expense of computation time.
   * When calculating prime implicants using `espresso`, instead of minterms pass a list of state-graph arcs and states to be covered.

2. At the level of factorisation, when using SIS:
   * Before factorisation explicitly represent inversion of a binate signal `x` as a new `x_bar` signal.
   * After factorisation restore the representation of `x'`.

3. At technology mapping, that is based on Boolean matching:
   * For binate signal `x` explicitly insert `x_bar=x'` and implement it as a zero-delay inverter.

**Main problems**:
* The main data exchange format in Petrify (e.g. result of minimisation) is BDDs where consensus cubes disappear.
* Source code (especially for technology mapping implemented by Enrico) is non-trivial and not documented. 
* It would take Jordi several weeks to implement this in Petrify, much longer for a new developer...

### Option 3

Do post-processing of the synthesised/mapped circuits and insert consensus cubes there (manually or partially automated).

1. Verify that implementation of signal with binate input has all the necessary cubes(see the sketch of the Reach expression above).

2. If a consensus cube is missing in implementation of signal `x` then:
* Indicate this to the user and rely on him to fix the problem manually.
* Calculate consensus cubes for `x` (somehow -- subject for a separate research) and insert them in the circuit, e.g. as an OR of the 'x' and the cubes (with timing assumptions).

vkhomenko, 2018-01-09:
# Verification check if a binate function implementation is problematic
Suppose a function `f`, that is binate in variable `x`, is implemented by a gate with function `G` in such a way that:
 * `G` has an input pin `p` connected to `x` (if `x` is connected to several pins of the gate, these pins are substituted by the same variable `p` in function `G`).
 * `G` has an input pin `n` connected to the output of the 0-delay inverter with the input `x` (if the inversion of `x` is connected to several pins of the gate, these pins are substituted by the same variable `n` in function `G`).
 * `G` can also take some other signals whose names are different from `x`, `p`, and `n`.

The property is checked on the composition of the circuit with the environment. It is violated in a state `s` if the following conditions are satisfied:
 1. There is an arc `s-x->s'`, i.e. a transition of `x` (with any sign) is enabled at `s`.
 2. `f(s)=f(s')`, i.e. `f` is insensitive to `x` in `s`, i.e. the derivative of `f` wrt. `x` in state `s` is 0: ~f'x(s). Alternatively, not to bother with `f`, one can replace this condition with `G[p:=1,n:=0](s) = G[p:=0,n:=1](s)`
 * `G` has a glitch on this transition, i.e. the derivative `G'p[n:=~x](s)=1`.

## Example
Compare the two implementations of the latch:
 * [Q]=C D + ~C Q
 * [Q]=C D + ~C Q + D Q
In both cases, `f = C D + ~C Q`, the binate variable is `C`, and the derivative is `f'C=D^Q`.

In the first case, `G = p D + n Q`, the derivative `G'p=(D + n Q)^(n Q)=D (~Q + ~n)` (note that the simplification step was optional), and `G'p[n:=~C]=D (~Q + C)`. Hence we verify the following Reach property that will fail and yield a trace to `s`:
```
@S"C" &
~($S"D"^$S"Q") &
$S"D" & (~$S"Q" | $S"C")
```

In the second case, `G = p D + n Q +D Q`, the derivative `G'p=(D + n Q +D Q)^(n Q +D Q)=D ~Q`  (note that the simplification step was optional), and `G'p[n:=~C]=D ~Q`. Hence we verify the following Reach property that will pass:
```
@S"C" &
~($S"D"^$S"Q") &
$S"D" & ~$S"Q"
``` 
## Derivation of consensus cubes
Note that the difference between `f'x` and `G'p[n:=~x(s)]`, i.e. `f'x & ~G'p[n:=~x(s)]`, is a function whose cubes may be candidates for consensus cubes.

PR #1039 (2019-11-08): Verify circuit implementation of binate functions
Closes #751
commit d185a8a5 (2019-11-08): Verify circuit implementation of binate functions

Closes #751
closed 2019-11-08 by danilovesky