workcraft/workcraft #757: Checking redundancy of places

vkhomenko, 2017-11-09:
The user should be able to select several places and verify that they can be removed from the net without affecting its behaviour. Also, for a single place this feature should be available via right-click.

vkhomenko, 2017-11-09:
The property could be as follows - but it needs some testing. 

- It should be checked with the -F (rather than -Fs) mode of MPSAT. 
- It is both for PNs and STGs.
- If the provided set of places is empty then the check is vacuous - but this is a dangerous situation so I think the front-end should disallow it.
- If the preset of some transition is included into the set of places passed to the check then the behaviour might be strange - I need to think about this situation carefully, so please don't release the tool until it's sorted out.
- Not sure about read arcs either.

````
// Checks whether the given set of places can be removed from the net
// without affecting its behaviour, in the sense that no transition can be
// disabled solely because of the absence of tokens on any of these places

let
  PNAMES = { "",
  	"p0","p2"
  } \ {""},
  PL = gather pn in PNAMES { P pn } {
    exists t in TRANSITIONS {
      ~@t
      &
      forall p in pre t \ PL { $p }
    }
}
````

danilovesky, 2018-01-18:
@vkhomenko The command is almost ready, just need to handle read-arcs and that "preset of some transition" situation. Do you have any advise about these?

vkhomenko, 2018-01-18:
I think the property is OK:
 * if the set includes pre t then after removing the places t is always enabled; the predicate in this case becomes ~@t, i.e. we check if t was always enabled in the original net
 * read arcs seem on too

commit 451ebc2f (2018-01-24): Command for checking redundancy of places

Closes #757
closed 2018-01-24 by danilovesky