workcraft/workcraft #772: Workcraft Crashes on MacOS

Ariela1127, 2017-11-24:
1. What version of Workcraft are you using?
Version 3.1.7

2. What operating system are you using?
MacOS High Sierra 10.13.1

3. What did you do? If possible, provide a list of steps to reproduce
   the error.
I created an STG model and tried to verify its properties via the Verification menu. For example, by clicking on the "Consistency [MPSat]" option.

4. What did you expect to see?
The verification steps executing.

5. What did you see instead?
Workcraft completely freezes and is unresponsive so I am unable to do anything. I can only force quit it via the main menu of the operating system (by clicking on the Apple logo). This issue happens every time I try to run any of the verification properties on my model.

danilovesky, 2017-11-26:
@Ariela1127 thanks for the report! I know of a similar problem when using JavaScript API, but it looks like you have this issue just using GUI, without any scripting...

Just in case, could you upload the `.work` you are having issue with (you may need to change the extension to `.zip` so GitHub does not complain)? Or does the problem persists for virtually any STG?

@jrbeaumont could you check if you can reproduce the problem on Mac using v3.1.7 and a basic STG (consistent and inconsistent) please? And if it is reproducible, please check if the latest changes in `master` resolve the issues.

jrbeaumont, 2017-11-26:
I have recently updated to MacOS High Sierra 10.13.1. But so far have not experienced issues with Workcraft, but let me to try to reproduce and see if we can help.

@danilovesky I have tested v3.1.7 with a consistent STG and and inconsistent STG, and verification completes successfully, and reports the correct result, with no crashes or hangs in Workcraft. Unfortunately I cannot reproduce this.

@Ariela1127 As above, please upload the `.work` you were using and I will try that. Did you notice any issues with Workcraft on previous versions of MacOS?

danilovesky, 2017-12-04:
It may be the case that Java update caused the issue. The problem was reported when Workcraft is run under Oracle Java version 8 update 151 (build 1.8.0_151-b12). @jrbeaumont could you try this version of Java please?

jrbeaumont, 2017-12-04:
I have tried this version, and still cannot reproduce, but I will keep an eye on it.

Ariela1127, 2018-01-18:
Does Workcraft keep a log file on the computer it was installed on? Everything worked fine for the past month, but now Workcraft sometimes freezes when I try to export an STG as a PGN file. There is nothing shown in the output window when it happens.

When I force quit Workcraft and try to export the same STG again, it works most of the time. I'm not sure if this issue is caused by my MacBook or the Java version that is installed, or something else completely.

danilovesky, 2018-01-18:
@Ariela1127 Unfortunately the only log is the one you see in the Output tab, it is not saved to the hard drive. But in any case, for image exporting we use Batik library (from Apache), so nothing to put in the log really -- it is just a call to some library function. I guess, similar to the previous issues, it would be very difficult to pinpoint the cause of the freeze :-/

danilovesky, 2020-10-09:
Closing as the issue did not manifest itself for a long time.

closed 2020-10-09 by danilovesky