workcraft/workcraft #858: Cannot cancel multi-threaded 64-bit external tool in Windows 10

danilovesky, 2018-07-20:
A multi-threaded 64-bit backend tool called from Workcraft cannot be cancelled in the Tasks tab.

If the cancel button is pressed while the backend tool is running the button text changes to "Cancelling...", it becomes inactive and stays like this forever.

Note that Linux binary of the same backend tool can be successfully cancelled, so it is something Windows-specific.

danilovesky, 2018-07-20:
This may be the cause of the problem: https://bugs.java.com/view_bug.do?bug_id=4770092

Here is a quote:

> The fundamental problem here is that, unlike Unix, Windows does that maintain parent-child relationships between processes. A process can kill its own immediate children, but unless you make other arrangements to obtain the information, can't kill any 'grand-children' because it has no way of finding them. Ctrl-C types at a command prompt is just a character that the command processor interprets and not a signal sent from outside. When you 'destroy' a child command script, that process does not get the opportunity to terminate any child processes it may know about.
> 
> Recent versions of WIndows (2000 or later) do provide a "Job" concept which acts as a container for processes. Killing a Job does terminate all processes associated with that job. However Jobs do not contain other jobs, so fully emulating the Unix behaviour is probably impossible.

vkhomenko, 2018-07-20:
Another interesting discussion: https://coderanch.com/t/384366/java/destroy-stop-Process-object

"neglecting to read output will cause a process to hang around indefinitely"

vkhomenko, 2018-07-20:
Actually, the output does not seem to be the problem in this case - just tried low verbosity mode and it still cannot cancel.