workcraft/workcraft #868: Improve deadlock verification for circuits

danilovesky, 2018-08-05:
Currently a circuit deadlock may be reported when the circuit does not conform to its environment. This happens because the environment does not accept the output events from of the circuit. This is counter-intuitive, as the circuit can still progress, and it is the environment that blocks the progress. It would be good to alter Reach property for circuit deadlock verification, so it takes this situation into account.

vkhomenko, 2018-08-06:
I think we discussed the mitigation, and sort of agreed not to interfere with the Reach property, but rather do the following:
  * when checking all properties, the deadlock should be checked last, and only if conformation succeeds;
  * when checking deadlock specifically, if a violation is detected, the message should say that it may be spurious if conformation is violated.

Actually, in the latter case one can be smarter and play the trace and see if there is an actual deadlock (i.e. the circuit cannot fire any gate); if the deadlock is spurious, the same trace is a witness of conformation violation, and can be reported as such; otherwise an actual deadlock exists.

danilovesky, 2018-08-06:
@vkhomenko If I am not mistaken we decided to do that quick fix (a separate ticket #867) and think about a better solution (possibly bringing part of Conformation property in). That solution with playing the trace may be a good one.

vkhomenko, 2018-08-06:
We did discuss bringing part of Conformation property in, but I think it's a bad idea. We then discussed simply telling the user that the deadlock may be spurious if conformation is violated - you assured me it's easy to edit the message.

danilovesky, 2018-08-06:
Closing the ticket then? Or converting it into "play the trace and check if the deadlock is real"?

danilovesky, 2018-08-12:
@vkhomenko I have changed the order of checks, so conformation goes before deadlock freeness.

Also if deadlock freeness is verified separately then all the reported traces are checked for conformation (so no outputs are enabled after the trace). If a trace leads to non-conformation the following warning ids logged in the Output tab after the trace:
``[WARNING] Deadlock trace is spurious because it leads to non-conformant output 'out'``

If all the reported traces lead to non-conformation, then the final message (also in a warning dialog) is as follows:
``[WARNING] Deadlock cannot be reliably checked because of conformation violation``

In case if some of the traces lead to a deadlock then these traces are reported for simulation, as it used to be before.