workcraft/workcraft #878: Structural checks for WTG before conversion to STG

danilovesky, 2018-09-06:
The following sanity checks can be performed:
- [ ] Every waveform has an entry state and exit state
- [ ] The First transition in every waveform as an input rise or fall event
- [ ] Some checks for the guards?
...

albertomv, 2018-09-07:
A signal must be stable before reaching a guard

- [ ] Guards must cover all options (and only once)
- [ ] Check that signal names don't collide with automatically generated names?
- [ ] Output signals should not be guards/become unstable
- [ ] In a waveform following a choice, the first signal cannot be an output (maybe never?)

albertomv, 2018-09-10:
- [ ] Destabilise and stabilise at 1/0 transitions cannot trigger an output

commit 398c2223 (2018-09-19): Added more structural checks

Closes #878
closed 2018-09-19 by danilovesky