workcraft/workcraft #902: Output-detrminacy check

vkhomenko, 2018-10-07:
- Before composition, need to check that there are no internal signals (as it is not clear whether to treat them as dummies or outputs), so either error or ask the user to choose which.
- The STG is then composed with itself.
- After the composition, phantom transitions are created in the composed STG for each output transition of the first component; their names are passed to the REACH property below. Actually, it's identical to the one for N-way conformation, though the composition is constructed differently!
- MPSAT needs to be modified to allow the "mixed" mode with the enabled-via-dummies semantics for @ but configurations with maximal dummies allowed.

// Check whether an STG is output-determinate.
// The enabled-via-dummies semantics is assumed for @.
// Configurations with maximal dummies are assumed to be allowed.
let
  // set of phantom output transition names in the whole composed STG
  PHANTOM_OUTPUT_TRANSITIONS_NAMES = {"", "x+/1", "x-", "y+", "y-/1"}\{""},
  PHANTOM_OUTPUT_TRANSITIONS = gather n in PHANTOM_OUTPUT_TRANSITIONS_NAMES { T n }
{
  // Optimisation: make sure phantom events are not in the configuration
  forall e in ev PHANTOM_OUTPUT_TRANSITIONS \ CUTOFFS { ~$e }
  &
  // Check if some output signal in the composition is enabled due to phantom transitions only;
  // this would mean that in the original STG there are two executions with the same visible
  // traces, leading to two markings M1 and M2 such that M1 enables some output o
  // but M2 does not enable o, so the output-determinacy is violated
  exists o in OUTPUTS {
    let tran_o = tran o{
      exists t in tran_o * PHANTOM_OUTPUT_TRANSITIONS{ @t }
      &
      forall tt in tran_o \ PHANTOM_OUTPUT_TRANSITIONS { ~@tt }
    }
  }
}

vkhomenko, 2018-11-10:
Another thing: If the STG is deterministic, i.e. there are no dummies and there are no choices between transitions labelled by the same output, then output-determinacy vacuously holds. Hence I suggest to do the same as we do for mutex protocol: Skip the check and report that it vacuously holds.

danilovesky, 2018-11-11:
I have opened this as a separate enhancement issue #923

commit 75f7b4e4 (2019-01-14): Revised check for output determinacy

* Structural check for vacuously held output determinacy -- closes #923
* Treat internal signals as outputs -- closes #902