workcraft/workcraft #957: Support for strict and relaxed mutex protocols

danilovesky, 2019-02-04:
Currently mutex is assumed to follows _strict_ protocol and never have both grants high. This, however, is too restrictive as some mutex implementations allow a new grant to be issued while the other grant is still being reset -- _relaxed_ protocol.

Strict mutex:
```
g1 = r1 * !g2 + g1 * r1
g2 = r2 * !g1 + g2 * r2
```
Relaxed mutex:
```
g1 = r1 * (!g2 + !r2) + g1 * r1
g2 = r2 * (!g1 + !r1) + g2 * r2
```
Both strict and relaxed mutex protocols should be supported, with the current one being selected via global preferences. This affects _Mutex place implementability [MPSat]_ command and also insertion of mutex after synthesis (correct set/reset functions need to be assigned).

vkhomenko, 2019-02-04:
If physical mutex element is assumed to be strict, then it can implement both strict and relaxed (and even mixed) protocols, and the REACH property in #943 works for this mutex.

If the physical mutex element is relaxed, the set and reset functions are as follows:
```
g1_SET=r1 & (~r2 | ~g2)
g1_RESET=~r1

g2_SET=r2 & (~r1 | ~g1)
g2_RESET=~r2
```
The next-state functions are as follows:
```
[g1]=r1 & (~r2 | ~g2)
[g2]=r2 & (~r1 | ~g1)
```
It can implement only the relaxed protocol, and so the REACH expression could be as follows:
```
// For given signals r1, r2, g1, g2, check whether g1/g2 can be implemented
// by a RELAXED mutex with requests r1/r2 and grants g1/g2.
// The properties to check are:
//   nxt(g1) = r1 & (~r2 | ~g2)
//   &
//   nxt(g2)=r2 & (~r1 | ~g1)
// Furthemore, the mutual exclusion of the critical sections is checked:
// ~( (r1&g1) & (r2&g2) )
// Note that the latter property does not follow from the above constraints
// for the next state functions of the grants (e.g. in the initial state).

let
  r1s = S"r1",
  g1s = S"g1",
  r2s = S"r2",
  g2s = S"g2",
  r1 = $r1s,
  g1 = $g1s,
  r2 = $r2s,
  g2 = $g2s,
  g1nxt = 'g1s,
  g2nxt = 'g2s
{
  (g1nxt ^ (r1 & (~r2 | ~g2)))  // negated definition of nxt(g1)
  |
  (g2nxt ^ (r2 & (~r1 | ~g1)))  // negated definition of nxt(g2)
  |
  r1 & g1 & r2 & g2  // mutual exclusion of critical sections
}
```

danilovesky, 2019-02-21:
Two global settings have been added to define which mutex protocol to use: 
  * `Digital Circuit -> Mutex protocol for synthesis` -- for synthesis with MPSat, Petrify and ATACS
  * `MPSat verification -> Mutex protocol for implementability check` -- for verification with MPSat.

This is partially for technical reasons (to avoid cyclic dependency between MPSat synthesis plugin and Digital Circuit plugin.

vkhomenko, 2019-02-21:
I'm worried about potential inconsistency if these two settings have different values - e.g. the user may change one of them and not being even aware of the other. Maybe these can be made one setting somehow?

danilovesky, 2019-02-21:
Initially I wanted to put this setting under a single ``Digital Circuit -> Mutex protocol``. However MPSat verification plugin cannot "use" Circuit plugin as it creates a cyclic dependency.

Another option is put it under ``Signal Transition Graph -> Mutex protocol`` (this is actually where the mutex protocol classes are), but this may also be confusing for the user. Or is it?

vkhomenko, 2019-02-21:
`Digital Circuit -> Mutex protocol` seems the right place as we are talking about the physical mutex. Note that many models might potentially refer to it, e.g. one might expand the WTG to work with mutexes, or state graphs. I think for now some sort of a quick fix would work, but in future one could create a "physical layer" meta-level that contains mutex info, WAIT-family stuff, gates info, etc., and is accessible by circuit plugin, STG plugins, etc.

snowleopard, 2019-02-21:
Some serious meta-level modelling is going on here: we will soon need to model Workcraft plugins using Petri Nets, eventually switching to self-hosting Workcraft in Workcraft :-)

danilovesky, 2019-02-21:
> Some serious meta-level modelling is going on here: we will soon need to model Workcraft plugins using Petri Nets, eventually switching to self-hosting Workcraft in Workcraft :-)

Actually we do already model some Workcraft stuff with Petri nets :-/

commit 73ef9e15 (2019-03-01): Support for strict and relaxed mutex protocols

Closes #957
closed 2019-03-01 by danilovesky