workcraft/workcraft #961: Compilation error JDK 9+ because of java.lang.Module

lwkbuit, 2019-02-09:
1. What version of Workcraft are you using?
v3.2.2/master

2. What operating system are you using?
Ubuntu 18.04

3. What did you do? If possible, provide a list of steps to reproduce the error.
./gradlew assemble

4. What did you expect to see?
No compilation errors

5. What did you see instead?

 > Task :PetriPlugin:compileJava FAILED
/home/luuk/git/workcraft/PetriPlugin/src/org/workcraft/plugins/petri/PetriNetModule.java:9: error: reference to Module is ambiguous
public class PetriNetModule implements Module {
                                       ^
  both interface org.workcraft.Module in org.workcraft and class java.lang.Module in java.lang match

danilovesky, 2019-02-11:
Thanks for spotting this!

Interestingly I have Java 8, 9, 10 and 11 installed in my Ubuntu and I run Workcraft with each of them from time to time, but never faced this issue. It appears that even though I switch Java version with `update-alternatives --config java` (and `java -version` confirms the desired version is active), Gradle is still using Java 8 for the build on my computer. I guess this is because of `sourceCompatibility = '1.8'` line in `gradle.build`.

@lwkbuit do you have any experience of how other projects deal with this name clash? I am thinking of these possibilities:

### Option 1
Use explicit interface name as `org.workcraft.Module`. This is an easiest solution, but would be difficult to enforce for Java 8 users.

### Option 2
Force Workcraft build to use Java 8 by add the following check to `gradle.build`:
```
   if(JavaVersion.current() != JavaVersion.VERSION_1_8){
        throw new GradleException("Java 8 is required for the build.")
    }
```

### Option 3
Rename the interface `Module` to something more distinguishable, e.g. `WorkcraftModule`.

danilovesky, 2019-02-11:
After reading a bit about name clashes in Java I decided to go with Option 1 for `org.workcraft.Module` (as the usage of this interface is rather localised) and with Option 3 for `org.workcraft.plugin.circuit.verilog.Module` (raneming it to `VerilogModule`).

lwkbuit, 2019-02-11:
Using fully qualified class/interface names is pretty ugly, but maybe you're right, the module interface is only used to enforce implementing the init/getDescription methods and doesn't appear much inline in code (with the PluginManager as an exception).

Personally I would choose Option 3.

commit 0e1f810f (2019-02-12): Add Verilog prefix to parser classes for Verilog

This is partially to deal with the name clash between
org.workcraft.plugins.circuit.verilog.Module and
java.lang.Module in Java 9 and newer.

Fixes #961
commit 2c365946 (2019-02-12): Use package qualification for Module interface

This is necessary for Java 9 and newer to distinguis
org.workcraft.Module from java.lang.Module

Fixes #961
closed 2019-02-12 by danilovesky