workcraft/workcraft #969: Integer properties not serialised

sefanja, 2019-03-18:
The `Integer` properties of my classes do not get serialised into the `.work` file. I have also tried this with `int`.

The fix that seems to work is to modify these lines:

https://github.com/workcraft/workcraft/blob/ddad08079aba93bd7510fd0022f8520ddc1e8e0f/WorkcraftCore/src/org/workcraft/plugins/builtin/serialisation/xml/IntSerialiser.java#L11

https://github.com/workcraft/workcraft/blob/ddad08079aba93bd7510fd0022f8520ddc1e8e0f/WorkcraftCore/src/org/workcraft/plugins/builtin/serialisation/xml/IntDeserialiser.java#L10

From this: `return int.class.getName();`
To this: `return Integer.class.getName();`

The first returns `int` while the second returns `java.lang.Integer`.

After this modification my `Integer` properties get (de)serialised.

sefanja, 2019-03-18:
Perhaps it's a best practice to use `int` instead of `Integer` on our plugin's classes? In that case it took me a long time to figure this out. ;)

danilovesky, 2019-03-18:
@mechkg By any chance do you remember the reason for serialisers/deserialisers to get the ClassName from `int`/`double`/`boolean`, and not `Integer`/`Double`/`Boolean`?

sefanja, 2019-03-18:
I'm relatively new to Workcraft, so unfortunately I don't know. I also just learned that using primitives over wrappers is a general best practice for Java, so that might have been the reason.

Originally I created this issue since I though that `int`s also didn't get serialised but that was due to an error in my code. So this is not an issue for me anymore, however it might help future plugin developers to know that wrappers don't get serialised (or to adjust Workcraft to serialise wrappers as well).

But for me personally, this issue has been resolved, now that I know to use primitives.

mechkg, 2019-03-18:
In Java `int` and `Integer` are two very different things, the former is a C-like primitive value whereas the latter is a Java object wrapping an `int` value. `int`-s are much faster and memory-efficient but can't be used in generic types (e.g. you can't have a List<int>).

I guess we simply decided that boxed primitives were rather rare as class fields so we just ignored that case, but there is no reason not to support them.

sefanja, 2019-03-18:
I've switched to `int`, works perfectly.

danilovesky, 2019-03-18:
@mechkg Thanks, this makes sense. I will reject the issue current collection of builtin serialisers/deserialisers seem to be good enough.

Another possible "benefit" of `Integer` over `int` is that it is nullable, so one can potentially use it as lightweight Optinal (Kotlin style) :)

closed 2019-03-18 by danilovesky