workcraft/workcraft #1108: 1-way conformation with dummies

danilovesky, 2020-07-03:
Improve the verification command for 1-way conformation so it accepts dummies. For this employ the shadow transition technique, similar to #901.

vkhomenko, 2020-07-03:
Note that we do 1-way conformation of circuits to their environments, with a guarantee that there are no dummies in the device STG. This is the same algorithm, so I hope you can re-use the code.

danilovesky, 2020-07-07:
@vkhomenko Just to clarify, would you prefer to:

A) preserve the current Reach expression for circuit conformation (which is guaranteed to be dummy-free) that uses a list of "device places" technique;

B) replace all conformation-related checks to use "shadow transitions" technique (even if there are no dummies in the STGs),

I am working towards option B)...

PR #1110 (2020-07-10): Support 1-way conformation with dummies
Closes #1108
commit 2be3a3dc (2020-07-10): Support 1-way conformation with dummies

Closes #1108
closed 2020-07-10 by danilovesky