workcraft/workcraft #1136: Reuse Worcraft instance when openning .work files from file browser

danilovesky, 2020-10-05:
Currently when opening .work files from a file browser a new Workcraft instance is started every time. A better option would be to re-use the already-running instance of Workcraft and open the .work files in its new tabs. Ideally this needs to be implemented in OS-independent way. Otherwise special cases would have to be considered for Linux, Windows and OSX.

ARandomOWL, 2020-10-05:
I don't think this is specific to file browsers. Maybe the first question is: do you want to support multiple instances of Workcraft, or allow only one?

Then we can think about how we handle the various scenarios: opening from app menu; terminal; opening a work file from a file browser.

danilovesky, 2020-10-05:
I tried to do this as follows: start a new instance of Workcraft and check if another instance is already running. If so, pass the opening request to that instance. Both these steps (checking and passing) required some OS-specific hacking. Maybe there is a cleaner way though...

ARandomOWL, 2020-10-05:
What did you try already? I was thinking about sockets.

danilovesky, 2020-10-05:
I tried lock files (with PID) and read about sockets. That only worked for detecting another instance. Did not manage to pass control to that instance in both Linux and Windows (never tried OSX).

danilovesky, 2020-10-08:
@ARandomOWL I Added `-port:<PORT>` parameter to Workcraft to specify the socket port for IPC. I am reluctant to put it in the runner script, as a hard-codded port may be taken on a specific PC (this would break the flow and the user would be puzzled of what is going on). So, this feature is for expert users for now, who know what they are doing - they can add the port parameter in their file association. For example, I have it as follows in Midnight Commander: `workcraft -port:12345 %p`. Is is a good enough solution?

closed 2020-10-09 by danilovesky

ARandomOWL, 2020-10-10:
If the port is already taken by another service, Workcraft still tries to "reuse the existing instance" and terminates. I guess this is expected for now, and up to the user to avoid.

danilovesky, 2020-10-11:
@ARandomOWL Yes, that port clash problem is the main reason I did not add it to the Workcraft starter script. Because otherwise a situation is possible when Workcraft does not start at all and the user would be puzzled of what is going on.

For now it is responsibility of the user to provide a "good" port number (as a positive side-effect there may be several Workcraft instances at their dedicated ports).

In the future I was thinking about some handshake between the client and the server sides, to make sure the port is actually listened to by a Workcraft instance. If the handshake fails a new instance of Workcraft would start.

Does this sound reasonable?

vkhomenko, 2020-10-11:
The h/s could exchange a (static) UUID.

PR #1137 (2020-10-09): Reuse running instance to open files from browser
Closes #1136
commit ba71025c (2020-10-09): Reuse running instance to open files from browser

Closes #1136