workcraft/workcraft #1144: Export PNG with background colour

ARandomOWL, 2020-10-19:
When exporting work as a PNG, the background is transparent. Usually I am exporting in PNG to share with others via messaging apps. If the recipient has a dark theme, the work becomes unreadable due to low contrast.

Perhaps there can be an option to export PNG with background colour or with transparent background?

danilovesky, 2020-10-19:
This can be easily changed by adding the following hint to `BatikUtils.transcode` method:
```
transcoder.addTranscodingHint(
        ImageTranscoder.KEY_BACKGROUND_COLOR,
        EditorCommonSettings.getBackgroundColor());
```

The challenge is to make it "optional" in the user interface. I see the following alternatives:
* Option A: Modify the file save dialog (currently a standard one is used) and add _Transparent background_ checkbox there (requires a lot of effort)
* Option B: Move this transparency choice to global preferences, .e.g. under _Common->Visual_ (easiest to implement)
* Option C: Have separate entries for opaque and transparent exports under _File->Export_ menu (may overcrowd the menu too much)

@ARandomOWL @vkhomenko Any preferences?

danilovesky, 2020-10-19:
Looks like the background colour hint only make difference for PNG export. All the vector graphics formats render the same with and without this hint. Therefore I will proceed with Option C:
//File->Export->Portable Network Graphics (*.png) - opaque background//
and
//File->Export->Portable Network Graphics (*.png) - transparent background//

Let me know if you see any issues with this..

vkhomenko, 2020-10-19:
Adding an extra menu item for a rarely used feature may be not the best solution - maybe Option B?

danilovesky, 2020-10-19:
@vkhomenko Which one do you think is "rare" -- transparent or opaque PNG?

vkhomenko, 2020-10-19:
Depending on the user, one of them is definitely rare :-)

I've been using the current setup (ok to insert into Word), but @ARandomOWL said, "Usually I am exporting in PNG to share with others via messaging apps. If the recipient has a dark theme, the work becomes unreadable due to low contrast", so it's the opposite for him.

vkhomenko, 2020-10-19:
BTW, since the Word started supporting SVG, both menu items are likely to become "rare" for me.

danilovesky, 2020-10-19:
Ok, now I leaning towards Option B  with the following setting under _Common->Visual_:
_Transparent background for PNG export_ (off by default).

vkhomenko, 2020-10-19:
You mean ON by default? to match the current setting?

danilovesky, 2020-10-19:
I would say transparent PNG is not very useful, so planned to have it OFF by default.
However, compatibility with the current setting is also a good reason to have it ON.
I cannot choose :-/

vkhomenko, 2020-10-19:
Ok, how about allowing to choose the background colour? ;-)
At least, in some future version.
Then transparent seems a reasonable default colour.

danilovesky, 2020-10-19:
That may be a bit too much :)
We already have a setting for editor background.
I would just use that for non-transparent PNG export.

vkhomenko, 2020-10-19:
Well, we then should use it always, including SVG export. Maybe worth making the background transparent by default?

danilovesky, 2020-10-20:
Ok, I did a colour option (transparent by default) _Background color for PNG export_.
All other formats are vector and do not have background (I do not plan to draw a coloured rectangle underneath).

commit dc2d47ba (2020-10-20): User-defined background color for PNG export

Closes #1144
closed 2020-10-20 by danilovesky