workcraft/workcraft #1168: Bug Report 14/12/2020

RoryBaker, 2020-12-14:
Please answer these questions before submitting your issue. Thanks!

1. What version of Workcraft are you using?
Version 3.3.1
2. What operating system are you using?
Window 10 Home
3. What did you do? If possible, provide a list of steps to reproduce
   the error.
I was working on a series of STGs and wanted to leave the program. I saved all the .work files I had open in Workcraft and pressed the red X in the top right hand corner to exit and close the program however it will not close. Workcraft had been open for several hours at this point
4. What did you expect to see?
The program to close and stop runnning.
5. What did you see instead?
The program will not close.

vkhomenko, 2020-12-15:
Dear Rory,

It seems difficult to reproduce the problem from this bug report:

- Is the problem in specific in 3.3.1, or it's also in 3.3.2?
- Were there any problems reported in the "Problems" pane?
- Were there any unfinished back-end processes running, like verification?
- Generally, is there a reliable way to reproduce the problem so that the issue can be debugged?

Thanks,
Victor.

RoryBaker, 2020-12-16:
> ⚠ External sender. Take care when opening links or attachments. Do not provide your login details.

Dear Victor,
It seems specific to 3.3.1, it has not yet occurred in 3.3.2 after I updated it.
There was nothing running at the time I tried to exit the program.
I could not repeat it on command however it has happened several times running it on the same computer.
Sorry that this isn’t very conducive to finding the bug.

Thanks,
Rory.