workcraft/workcraft #1169: Inplace editor fails to render the text correctly

TheCodeSummoner, 2020-12-16:
1. What version of Workcraft are you using?

_Workcraft 3 (Return of the Hazard), version **3.3.2**_

2. What operating system are you using?

Linux 64-bit:
```
Description:	Manjaro Linux
Release:	20.2
Codename:	Nibia
```

3. What did you do? If possible, provide a list of steps to reproduce the error.

Created a new text box and started typing. Then attempted to modify some text by selecting it with the cursor.

4. What did you expect to see?

The text being updated correctly.

5. What did you see instead?

Rendering issues.

---

See the below image:

[image]

This might be connected to #209 - rather than just changing the bounds dynamically, perhaps some enhancements to text rendering itself should also be implemented.

cc @vkhomenko (mentioned this issue in the e-mail)

ARandomOWL, 2020-12-16:
Unable to reproduce
Workcraft 3.3.2
Manjaro Gnome 20.2 Stable
jre-openjdk 14.0.2.u12-1

@danilovesky Could this be a font issue? I don't see any preferences for font choice, so not sure which one is being used.

vkhomenko, 2020-12-16:
Works ok in Windows too.

Could it be some poor java library implementation for that OS, e.g. some unstable java release?

danilovesky, 2020-12-17:
I can reproduce it. The text needs to be rather long and after double-clicking you need move the text cursor around.

ARandomOWL, 2020-12-18:
> I can reproduce it. The text needs to be rather long and after double-clicking you need move the text cursor around.

I see it now. It only happens at certain zoom levels.

danilovesky, 2020-12-18:
@ARandomOWL I was suspecting zoom levels that result in non-integer size of the font. However, I can still reproduce it even at restricted zoom levels tight to integer font size. I am puzzled and not even sure who to blaim, as this is just a standard JTextPane component that displays the text.

danilovesky, 2021-09-15:
The problem is related to the use of Overlay as a parent for the text editor component.
A possible solution is to use JDialog as a parent for the text editor and position it relative to the GraphEditor canvas.

commit 82b94bd2 (2021-09-16): Improve in-place editor using JDialog

Fixes #1169
closed 2021-09-16 by danilovesky