workcraft/workcraft #1225: Incorrect detection of place redundancy in presence of read arcs

danilovesky, 2021-04-21:
**To reproduce:**
Check redundancy of place `p_test` (highlighted in magenta) in the following STG:
[image]

**Observed behaviour:**
The place is redundant (unexpectedly) .

**Expected behaviour:**
The place is essential as it prevents `out+` from firing in the initial state (without `p_test` it becomes enabled).
 
Note: The above example can be downloaded here (remove .zip extension): [redundant_place.stg.work.zip](https://github.com/workcraft/workcraft/files/6352754/redundant_place.stg.work.zip)

-----

The reason for this unexpected behaviour is due to `-r` option that is passed to Punf by default. With this option Punf splits places with read-arcs and renames the created copies. In this case we end up with `p_test` and `p_test@1`, of which the former is indeed redundant (has the same adjacent arcs as `p1`), while the later is essential.

A solution is to alter Reach expression, so it identifies the replicated places using a regular expression:
```
// Checks whether the given set of places can be removed from the net without affecting its behaviour, in the
// sense that no transition can be disabled solely because of the absence of tokens on any of these places.
let
    PNAMES = {"ready", ""} \ {""},
    // find places with names in PNAMES, possibly with suffix @N
    PL = gather p in PLACES s.t. exists pn in PNAMES { p in PP (pn + "(@[0-9]+)?") } { p }
{
    exists t in post PL {
        ~@t
        &
        forall p in pre t \ PL { $p }
    }
}
```

vkhomenko, 2021-04-22:
Maybe a slightly better REACH expression:
```
// Checks whether the given set of places can be removed from the net without affecting its behaviour, in the
// sense that no transition can be disabled solely because of the absence of tokens on any of these places.
let
    PNAMES = {"ready", ""} \ {""},
    PLSS = gather pn in PNAMES { PP (pn + "(@[1-9][0-9]*)?") }, // set of sets of places
    PL = gather p in PLACES s.t. exists ps in PLSS { p in ps } { p } // the union of sets in PLSS
{
    exists t in post PL {
        ~@t
        &
        forall p in pre t \ PL { $p }
    }
}
```

It's annoying that it's not trivial to compute the union of a set of sets :-(

vkhomenko, 2021-04-22:
Another possibility: If the frontend could compute the regex, then the expression is much simpler:
```
// Checks whether the given set of places can be removed from the net without affecting its behaviour, in the
// sense that no transition can be disabled solely because of the absence of tokens on any of these places.
let
    PNAMES = "p1|p2|p3",
    PL = PP ("(" + PNAMES + ")(@[1-9][0-9]*)?")
{
    exists t in post PL {
        ~@t
        &
        forall p in pre t \ PL { $p }
    }
}
```

vkhomenko, 2021-04-22:
Formulation that does not use regex:
```
// Checks whether the given set of places can be removed from the net without affecting its behaviour, in the
// sense that no transition can be disabled solely because of the absence of tokens on any of these places.
let
    PNAMES = {"ready", ""} \ {""},
    PL = gather p in PLACES s.t. exists pn in PNAMES { name p=pn | (name p)[..len pn] = pn + "@" } { p }
{
    exists t in post PL {
        ~@t
        &
        forall p in pre t \ PL { $p }
    }
}
```

vkhomenko, 2021-04-22:
Alternative formulation that would generate an error if there are no places with given names:
```
// Checks whether the given set of places can be removed from the net without affecting its behaviour, in the
// sense that no transition can be disabled solely because of the absence of tokens on any of these places.
let
    PNAMES = {"ready", ""} \ {""},
    PL = gather pn in PNAMES { P pn } + 
            gather p in PLACES s.t. exists pn in PNAMES { (name p)[..len pn] = pn + "@" } { p }
{
    exists t in post PL {
        ~@t
        &
        forall p in pre t \ PL { $p }
    }
}
```

commit 19acb2ff (2021-04-22): Fix place redundancy check with read arcs

Closes #1225
closed 2021-04-22 by danilovesky