workcraft/workcraft #1226: Command to check if there is redundant place in a selection or in the whole model

danilovesky, 2021-04-21:
The user may need to check if any of the selected places is redundant (if no place is selected, then check the whole set of STG/Petri net places for redundancy). As an optional usability feature, report/highlight/select one of the redundant places (if any).

vkhomenko, 2021-04-22:
MPSAT can find and remove a maximal (w.r.t. set inclusion - this is different from max cardinality!) set of redundant places. In particular, dead places are always redundant. Note that when removing redundant places, one must be careful: Removing a dead place may make some dead transition alive (MPSAT removes dead transitions when removing redundant places).

This functionality is not directly accessible though. Maybe a new MPSAT mode is necessary for this.

vkhomenko, 2021-04-23:
This could be generalised to a check if there is any redundancy at all in the model, e.g.:
  * a place can be redundant (dead or implicit);
  * a read arc can be redundant (if the incident place is marked whenever the incident transition's other pre-places are marked);
  * a transition can be redundant (dead, duplicate of another transition, loop-only).