workcraft/workcraft #1235: In mutex protocol verification, do not consider the grants as triggers

vkhomenko, 2021-05-03:
Related to #682.

When determining the grants and requests in mutex protocol verification, first the grants are determined syntactically, by looking at the postset of the mutex place. Then requests are determined as syntactic triggers of these grants.

I think, after computing these triggers, one should remove the grants (if any) from the set of triggers before determining the requests. This is useful as there are practical situations when grants may syntactically trigger grants (via places other than the mutex place). 

As an example, consider the usual mutex STG (doesn't matter whether it's relaxed or strict protocol). Now, suppose we want to add some extra transition t with the guard g1=0. To do that, we would create a new initially marked place g1_LOW, and add arcs g1_LOW -> g1+ and g1- -> g1_LOW, and a read arc g1_LOW --- t. Unfortunately, this seemingly innocent transformation currently breaks the verification of mutex protocol, as now g1- becomes a syntactic trigger of g1+ (via g1_LOW) and so both r1 and g1 are considered as requests of g1.

Note also that if we ever add generalised STGs allowing guards on transitions, such a transformation will be automatically performed behind the screen, resulting in mysterious verification failures.

danilovesky, 2021-05-04:
This definitely makes sense for syntactic self-triggers (as for g1 in the above example). I am still not sure if this is the case for g2 being being a syntactic trigger of g1. Do you recon all grants should be removed from the list of triggers?

vkhomenko, 2021-05-04:
I believe both grants should be removed. Note that g2 is already a trigger of g1 in the strict protocol (via the mutex place), so probably there is no harm to make it a trigger via some other place. I don't have a convincing use case however, but such a case may exist.

Generally, I think the philosophy is that the frontend guesses what the requests and grants are, and if the subsequent verification passes, we call it a success. So the frontend should try to fail making a guess as rarely as possible. Which means removing the grants from the set of triggers is a good thing.

Another idea: Suppose we are looking for the triggers of a transition t (e.g. t=g1+), and suppose p is in *t. It may happen that there is some potential trigger t' in *p, which is connected to p by a read arc. This means t' cannot be a real trigger (unless it triggers t via some other place), so it should be excluded. Formally, we only consider the set strict_pre(*t) as potential triggers, where strict_pre is preset excluding read arcs, and we further remove the grants from this set.

commit e0848546 (2021-05-07): Skip mutex grants when detecting their syntactic triggers

Closes #1235
closed 2021-05-07 by danilovesky