workcraft/workcraft #1253: Failure of CSC conflict resolution is not handled properly

danilovesky, 2021-06-23:
MPSat logic synthesis may fail due to unresolved CSC conflicts. This, however, is not handled properly, as the return code is 1 (interpreted as success alongside 0). This situation should be detected by checking `stderr` for the line `Warning: failed to resolve some of the encoding conflict`.

CSC conflict resolution with Petrify fails, but this is not reported to the Windows users due to incorrect handling of the new line characters when analysing `stderr` stream.

vkhomenko, 2021-06-23:
I'll have a look, maybe it makes sense to promote this to an error?

danilovesky, 2021-06-23:
No worries, I will detect it via stderr for now (and add few unit tests for irreducible CSC conflicts -- those were missing).
If eventually promoted error (which makes sense), I will clean up the code a bit.

PR #1254 (2021-06-23): Handle failures in CSC conflict resolution
Fixes #1253
commit 69ef5251 (2021-06-23): Handle failures in CSC conflict resolution

Fixes #1253
closed 2021-06-23 by danilovesky